Tlf/tlf #31: Bandmap improvements

dh5fs, 2015-01-17:
One of the big weak points for realtime logging is still the bandmap display. In detail:
- the shown number of stations is too small > bigger terminal window possible, or even a seperate one?
- no connection of the bandmap to the actual frequency > if the number of calls stays limited there should be some sorting around the actual working frequency
- no filtering for Mults possible, not even for DXCC Mults
- no free movement in the bandmap possible > something like ALT-ARROW UP, DOWN to move up/down in frequency would be nice

73 Fred

dl1jbe, 2015-09-19:
Actual state - a short statement:

> no connection of the bandmap to the actual frequency > if the number of calls stays limited there should be 
> some sorting around the actual working frequency

That is implemented now. See commit  
https://github.com/Tlf/tlf/commit/3be8dd518b84d1e00209e6ce840ad274a3342720

> free movement in bandmap

needs quite some work on key input handling as ncurses limits the range of usable key combinations. It is on the ToDo list, but will take time.

> display of more spots, bigger window

Some ideas to allow more than 25 lines for the program window are in the work at the moment. Unluckily most display postions are hard coded so it is quite some work to make it more flexible.

dl1jbe, 2015-10-06:
bandmap display is now centered around center frequency of actual band and mode segment if you have no running rig connected.

dl1jbe, 2015-10-13:
HA2OS provided a patch to save and reload bandmap data. That will be helpful if tlf crashes or you have to temporarily leave tlf.

zcsahok, 2020-01-13:
Fred, are there still something that need to be fixed/added or can we close this issue?

dh5fs, 2020-01-13:
Still to move with CTRL-G only into one direction and not to move in the opposite frequency direction if needed would be a valuable improvement.

zcsahok, 2020-01-13:
Do you mean not reversing search direction at the end of the list, but rather jump to the beginning?

dh5fs, 2020-01-13:
I mean to change direction whenever needed, not just at the end of the list.

zcsahok, 2020-01-13:
I see, makes sense. Just have to come up with a key for it. Maybe `Ctrl-Shift-G` could work.
Eventually mouse clicking on the list could be supported too. I saw plans for this.

zcsahok, 2021-01-07:
How about a grab direction indicator as shown in the screenshot?
Arrows (well, pretty crude) pointing downwards mean grabbing up in frequency matching the visual representation of the band.
One could either set the direction using say Ctrl-Up/Down arrows (but they are not available on console) or toggle it using some free key, e.g. `*`, tilde or backtick.

[image]

dh5fs, 2021-01-07:
Well, if for whatever reason a CTRL-UP/DOWN mechanism is not possible it would be a good workaround to have a "switch key" for the direction and a indication which direction is active.
73 Fred

dl1jbe, 2021-01-08:
Wrt Ctrl-Up/Down. That combination does only work with xterm. In urxvt and linux console we can not use it.

As an alternative idea we could use ALT_U and ALT_D for up and down. May be good to memorize.
Otherwise we can use the bandmap '.' menu and use just up and down arrow or a direction switch.

dl1jbe, 2021-01-23:
Please try the following branch and report back:
https://github.com/dl1jbe/tlf/tree/grab_up_down

Uses ALT_U/D to go up and down.

dl1jbe, 2021-01-23:
Sorry forgot to push the code. Now the link above is correct.

zcsahok, 2021-01-23:
Works in xfce term, but not on console.

Here is my draft with grab direction indicator and Ctrl-V toggling the direction:
https://github.com/zcsahok/tlf/tree/grab_dir

One anyway uses Ctrl-G to grab and Ctrl-V is just next to it. (and happens to be free, a rarity :-)  )

dl1jbe, 2021-01-23:
Weird. Console works great here. Maybe it is your local keyboard setting.
Is the problem on both keys or only one of it? Here both keys are mapped to Meta_d and Meta_u.

Anyway. I will have a look at your version tomorrow.

zcsahok, 2021-01-23:
Works, just didn't pull.
One has to get used to press _U_ to grab the spot _below_ the green mark. And when user gets back to Ctrl-G then it grabs in the original (kind of random, unrelated to previous Us/Ds) direction.

dl1jbe, 2021-01-23:
Ok, _Up_ in frequency vs. _up_ in list - sometimes people have different expectations :-).

ALT_U/D should remember the direction and let Ctrl-G continue in that direction. The question would be if we need Ctrl-G then anymore.

dh5fs, 2021-01-24:
Wow, now 2 options to choose from :+1: Out of tradition I would opt for CTRL-G + CTRL-V, but most other programs have rather the Up/Down mechanism.

dl1jbe, 2021-01-24:
If Ctrl-V works stable and ALT_U/D makes trouble lets stay with the first one. I am fine with both versions.

dh5fs, 2021-01-24:
Great, then we can concentrate now on advanced filtering options for the bandmap :laughing:

dl1jbe, 2021-01-25:
@zcsahok please provide a PR when you are done with your improvement. Will pick it up then.

zcsahok, 2021-01-27:
@dh5fs I suggest closing this issue (if the solution is OK) and opening a new one on further improvement options.

dl1jbe, 2021-02-06:
Problems wrt bandmap related segmentation faults got fixed in PR #239.
@dh5fs please retest and close Issue if successful. Please open new Issue one for further improvements.

dh5fs, 2021-02-06:
Thanks, all working fine now. Tested in EURASIA contest this weekend.

closed 2021-02-06 by dh5fs