Tlf/tlf #50: With DX_&_SECTIONS set in rules file, NE counted as mult instead of ONE

N0NB, 2015-12-06:
During the 2015 ARRL 160m contest from USA when a station in the ONE section was worked, NE was counted as the mult instead.  Seeing that NE appears in the file arrlsections before ONE, a change may need to be made where the length of the string in the file must match that entered in the exchange field.

More discussion in the thread with this message:

http://lists.nongnu.org/archive/html/tlf-devel/2015-12/msg00022.html

dl1jbe, 2015-12-14:
I found similar problems in the code for ARRL SS handling. The wrong handling is coming from the addmult() code. As I am at reworking it in the moment anyway I will assign the issue to myself but any suggestions and patches are welcome.

N0NB, 2015-12-14:
I will note that after manually editing the log file so the mults were noted correctly that upon restart Tlf calculated the score correctly along with the mults.

I was looking at the checkexchange function in getexchange.c and looking at the pattern sequences of serpats and the like, I wondered why regular expressions aren't used instead.  The glib library offers regex functions that are based on the pcre library.  Perhaps it wouldn't be an advantage as I'm still studying the flow.

airween, 2015-12-14:
Hi Nate,
there is a command (type in call field): ":rescore", and Tlf will rescore all your point with multis. Hope this helps.

73, Ervin

dl1jbe, 2015-12-14:
Nate, your observation after manually correcting the log is understandable. The mult scoring uses different code paths when you input a new qso and when you reread your logfile (after a restart or the :rescore Ervin noted.

For the first path look at addmult.c around line 100 where the code checks if it is a possible multi. The error here is that it accepts the first one found even if a better is in the list (but coming later in the multi file).

The second path comes via readcall.c and the readcalls() function. Here it uses the multis recorded in the log file and do not check again if it is one of the allowed multis. 

As you suggests using regular expression may be one possible solution for the problem.

N0NB, 2015-12-14:
* On 2015 14 Dec 14:37 -0600, Ervin Hegedus wrote:
> Hi Nate, there is a command (type in call field): ":rescore", and Tlf
> will rescore all your point with multis. Hope this helps.

Hi Ervin.

I did try that but no change.  Editing the log manually to get the
sections correct did work.  This is one I will be testing in the future.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Ham radio, Linux, bikes, and more: http://www.n0nb.us

dl1jbe, 2015-12-15:
As I see it, we need to look for the longest possible multipliers first. If they do not much we can test the shorter ones.

For that it seems to be enough to sort the possible multipliers by length before lookup.

dl1jbe, 2015-12-23:
In meantime pull request #60 should fix the problem. I will merge it in, if I get no negative feedback in next days.

dl1jbe, 2016-01-01:
A final comment to the question about the regular expressions raised above:

The checkexchange() function is still a very old remnant of the original code from Rein. And yes it should be rewritten using regular expressions.

PR #60 (2016-01-01): rework of addmult() function
The proposed patch provides some cleanup and error correction for loading an external multfile and registering multis. It contains the following changes:

- Drop not needed code and double checks for border conditions
- Use the existing 'remember_multi()' function also for arrl Sweep stake registration
- init_init_and_load_multipliers() (former load_multipliers() )  is now responsible for (freeing and new) allocation of the mults_possible array.
It now handles empty multplier filenames gracefully by giving back an empty mults_possible array. The former check for na [... cut]
closed 2016-01-01 by dl1jbe