Tlf/tlf #51: panel.h not found on some distributions.

N0NB, 2015-12-06:
Per this thread:

http://lists.nongnu.org/archive/html/tlf-devel/2015-12/msg00009.html

David reported an issue with his distribution not havng panel.h in /usr/include.  At this time configure.ac does not have an explicit test for the location of panel.h.  Suspect this can be resolved by use of extra macros available from the Autoconf macro archive:

http://www.gnu.org/software/autoconf-archive/

Tom, you may assign me to this, if you can.

dl1jbe, 2015-12-06:
Seems I can only assign organisation members. Lets see what can be done here.

dl1jbe, 2015-12-07:
So, now I could assign you.

N0NB, 2015-12-08:
I have submitted pull request #62 for this issue.

dl1jbe, 2015-12-17:
Closing the issue as pull request #56 solved the problem. Thanks Nate.

closed 2015-12-17 by dl1jbe