Tlf/tlf #57: search for ncurses fails with new ax_with_curses.m4 on some systems

dl1jbe, 2015-12-18:
After configure I got "SysV or X/Open-compatible header file not found" on some systems.

Further inspections shows:
- ncurses got installed with unicode support so /usr/include/ncurses.h and /usr/include/ncursesw/ncurses.h are installed.
- configure found the ncursesw/ncurses.h which are not supported in out tlf_ncurses.h
- The supplied ax_with_curses.m4 is patched to *not* look for ncursesw but anyhow it got found.

So finally:
- The system has a preinstalled ax_with_curses.m4 which is not patched and gets used instead of our file.

Some possible ideas for getting around:
- patch the supplied ax_with_curses.m4 and rename the helper function to a tlf specific one. (Ugly solution)
- See if we can control the with_ncursesw and with_ncurses variables from outside the m4 file 

Other ideas?

N0NB, 2015-12-19:
By default that macro will search for ncursew before ncurses.  If this is not a problem, and if we'd like to enable the use of "wide" characters in the future, then the easiest is to probably unset the variables I set in the macro file and let it determine which library is available.

As I understand it, ncursesw is backward compatible to ncurses, but I chose to err on the side of caution by only enabling ncurses and curses.

As for the options, I *think* they can be set in configure.ac and will be honored.  At the very least we should test them so that they're sane, but looking at the documentation I don't see a combination that would result in a situation where some curses library is never chosen.

Probably the easiest (best?) solution is to uncomment the changes I made to ax_with_curses.m4 and go with it.  :-)

dl1jbe, 2015-12-19:
Downside on simple using the original macro is that anytime ncursesw is selected we get the above error message.

So we either have to accept the wide character support or give a comment, e.g.
'At the moment tlf does not support wide character display, please use --without-ncursesw'

Both cases needs an correction of the tlf_[ncurses|panel].h files.

N0NB, 2015-12-19:
* On 2015 19 Dec 10:39 -0600, Thomas Beierlein wrote:
> Downside on simple using the original macro is that anytime ncursesw
> is selected we get the above error message.
> 
> So we either have to accept the wide character support or give a
> comment, e.g.  'At the moment tlf does not support wide character
> display, please use --without-ncursesw'

I am guessing the library is backward compatible.  I am also guessing
that since most everything is using UTF-8 these days, just sticking with
the ASCII character set is not a problem.

As I see it, I18n is down the road a ways, probably 2.0 or later.

> Both cases needs an correction of the tlf_[ncurses|panel].h files.

I'll send a pull request in a bit that touches all of these.  It should
be applied before the configure.ac stuff.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Ham radio, Linux, bikes, and more: http://www.n0nb.us

N0NB, 2015-12-19:
Submitted pul request #59 to address this and also added a comment to the commit.

dl1jbe, 2015-12-21:
Closed with pull request #59 merged in. Thanks.

closed 2015-12-21 by dl1jbe