Tlf/tlf #64: Ubuntu 14

timseed, 2016-01-22:
Ubuntu 14 seems to dislike the configure.ac 

I am getting lots of MACRO may not be found ....

Command I am giving is

    autoconf

airween, 2016-01-22:
Hi Tim,

what's the error message exactly?
Btw, you have to type these:

```
autoreconf --install
./configure --enable-hamlib --enable-fldigi-xmlrpc
make
sudo make install
```

(Note, that the --enable-hamlib and --enable-fldigi-xmlrpc are optional.)

Is this helped?

73, Ervin

timseed, 2016-01-22:
Hi there,
    I just pulled a fresh copy from github -untar-ed and then followed
your instructions

autoreconf --install
configure.ac:57: warning: macro 'AM_PATH_GLIB_2_0' not found in library
configure.ac:60: error: possibly undefined macro: AM_PATH_GLIB_2_0
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1

It does create configure - but if I try and use it then I get....

./configure
configure: error: cannot find install-sh, install.sh, or shtool in "."
"./.." "./../.."

OS Ubuntu 14 64 Bit. specifically (from uname -a)

Linux photo-lt 3.19.0-42-generic #48~14.04.1-Ubuntu SMP Fri Dec 18
10:24:49 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Regards

   Tim

timseed, 2016-01-22:
A solution seems to be to disable warnings....

    autoreconf --install  --warnings=none

Generates the output files... I am just installing the missing library's and then I can confirm if it is working or not.

timseed, 2016-01-22:
All working now.... a few missing library's - and the warnings=none was the issue.

It is running on Ubuntu 14.01 64 Bit

closed 2016-01-22 by timseed