Tlf/tlf #72: Problem compiling tlf due to ncurses

sebastiankricner, 2016-11-09:
Hi,

i am unable to compile tlf anymore with ncurses on gentoo:

checking whether pthreads work with -kthread... no
checking for the pthreads library -llthread... no
checking whether pthreads work with -pthread... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking if more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking for x86_64-pc-linux-gnu-pkg-config... /usr/bin/x86_64-pc-linux-gnu-pkg-config
checking pkg-config is at least version 0.16... yes
checking for x86_64-pc-linux-gnu-pkg-config... (cached) /usr/bin/x86_64-pc-linux-gnu-pkg-config
checking pkg-config is at least version 0.16... yes
checking for GLIB - version >= 2.0.0... yes (version 2.46.2)
checking for NcursesW wide-character library... yes
checking for working ncursesw/curses.h... no
checking for working ncursesw.h... no
checking for working ncurses.h... no
configure: WARNING: could not find a working ncursesw/curses.h, ncursesw.h or ncurses.h
configure: error: a Curses library with color support is required

!!! Please attach the following file when seeking support:
!!! /var/tmp/portage/media-radio/tlf-9999/work/tlf-9999/config.log
 * ERROR: media-radio/tlf-9999::tuxwave failed (configure phase):
 *   econf failed
 * 
 * Call stack:
 *               ebuild.sh, line  115:  Called src_configure
 *             environment, line 3565:  Called econf '--docdir=/usr/share/doc/tlf-9999' '--enable-hamlib'
 *        phase-helpers.sh, line  662:  Called __helpers_die 'econf failed'
 *   isolated-functions.sh, line  117:  Called die
 * The specific snippet of code:
 *   		die "$@"
 * 
 * If you need support, post the output of `emerge --info '=media-radio/tlf-9999::tuxwave'`,
 * the complete build log and the output of `emerge -pqv '=media-radio/tlf-9999::tuxwave'`.
 * The complete build log is located at '/var/log/portage/media-radio:tlf-9999:20161109-165623.log'.
 * For convenience, a symlink to the build log is located at '/var/tmp/portage/media-radio/tlf-9999/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/media-radio/tlf-9999/temp/environment'.
 * Working directory: '/var/tmp/portage/media-radio/tlf-9999/work/tlf-9999'
 * S: '/var/tmp/portage/media-radio/tlf-9999/work/tlf-9999'

>>> Failed to emerge media-radio/tlf-9999, Log file:

>>>  '/var/log/portage/media-radio:tlf-9999:20161109-165623.log'

There is ncursesw/curses.h.

airween, 2016-11-10:
Hi Sebastian,
as I see, this is a distribution build system output (may be you're using Gentoo?). I'm not a Gentoo user, so I can't help you exactly in this  problem - but could you try to build through the "native" mode? I mean, you grab the source/git repo, autoreconf --install, ./configure ..., make and make install?
You can pass several arguments to configure, eg:
`./configure CFLAGS="I/path/to/where/ncursesw/exists" --enable-hamlib`

a.

sebastiankricner, 2016-11-10:
Well, it does not seem to be due the buildsystem of gentoo. Your suggested command also failed with the same result.

So far:
` ./configure --without-ncursesw --enable-hamib`
works. Have you tested compiling tlf with ncurses wide char support? I suppose, it may happen, that developers are using ncurses without ncursesw support and compiling fine due to this.
Is tlf even able to use wide character?

So, i think, the autotools tests for ncurses need to be checked, whether they may pass if ncursesw is selected.

airween, 2016-11-10:
I've installed libncursesw5-dev (on my Mint), and the result is:

`checking for working ncursesw/curses.h... yes`
(and the Tlf had been build at the end of the all flow).

automake 1:1.14.1
autoconf 2.69
m4 1.4.17
*ncurses* 5.9

Could you share with me your config.log (when your configure script runs to fail)? You can send me to direct.

sebastiankricner, 2016-11-10:
I sent you a e-mail.
Also:

https://autotools.io/pkgconfig/pkg_check_modules.html
https://autotools.io/pkgconfig/index.html

Would be great for ncurses

dl1jbe, 2016-11-10:
Am Thu, 10 Nov 2016 08:33:16 -0800
schrieb Sebastian Kricner <notifications@github.com>:

> I sent you a e-mail.
> Also:
> 
> https://autotools.io/pkgconfig/pkg_check_modules.html
> https://autotools.io/pkgconfig/index.html
> 
> Would be great for ncurses
> 

Hi Sebastian,

you are true. I did not check it yet for ncursesw settings.

So, can you please send me your USE flags from make.conf, any possible
settings from package.use for ncurses and your tlf-9999.ebuild please
per pm.

As I run gentoo myself it should be not so hard to find out.

73, de Tom DL1JBE.

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

sebastiankricner, 2016-11-10:
Hi,

the problem is just, that the config tests were not successful due to missing -ltinfo.

pkg-config --libs ncurses reports:
-lncurses -ltinfo

I use tinfo use flag for ncurses, which means, that the tinfo library is built separately.
So the best way to handle it for systems, which use split ncurses/tinfo libraries, is to use pkg-config for the linker flags.

dl1jbe, 2016-11-10:
Am Thu, 10 Nov 2016 09:18:26 -0800
schrieb Sebastian Kricner <notifications@github.com>:

> Hi,
>
> the problem is just, that the config tests were not successful due to
> missing -ltinfo.
>
> pkg-config --libs ncurses reports:
> -lncurses -ltinfo
>
> I use tinfo use flag for ncurses, which means, that [...]

Well, I see. We had a pkg-config check before the last big change in
ncurses handling. There was a reason we stayed away from it. I have to
dig a little bit into history to find out what happened and what the
best solution maybe.

I will test it with USE=tinfo myself. Thanks for reporting.

Tom 

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

sebastiankricner, 2016-11-10:
Alright. The problem is solved locally. But i have read, that other distros are using separate tinfo/ncurses libraries, like Arch Linux.

In my oppinion, the macro AX_WITH_CURSES should handle it properly, as the page http://www.gnu.org/software/autoconf-archive/ax_with_curses.html states:

Finding libraries will first be attempted by using pkg-config, and should the pkg-config files not be available, will fallback to combinations of known flags itself.

dl1jbe, 2016-12-16:
Hi Sebastian,

Am Thu, 10 Nov 2016 09:55:06 -0800
schrieb Sebastian Kricner <notifications@github.com>:
> In my oppinion, the macro AX_WITH_CURSES should handle it properly,
> as the page
> http://www.gnu.org/software/autoconf-archive/ax_with_curses.html
> states:
>
> Finding libraries will first be attempted by [...]
That are the new AX_WITH_URSES macros which came out this year. That
indeed fixes it for most systems.

I just patched the Gentoo ebuilds and picked the new macros up in the
tlf/ncurses6 branch. It will become part of next tlf release.

Thanks,

	Thomas.

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

sebastiankricner, 2016-12-18:
That are good news, so i mark this issue as closed.

closed 2016-12-18 by sebastiankricner

dl1jbe, 2016-12-18:
Am Sun, 18 Dec 2016 12:35:02 -0800
schrieb Sebastian Kricner <notifications@github.com>:

> That are good news, so i mark this issue as closed.
> 

Thanks.

	Thomas

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--