Tlf/tlf #73: cwops cwt contest request

jg00dman, 2017-02-06:
Hi, I've read all the documentation I can find but I'm struggling to create a config file which can support the cwops cwt. I wonder if this could be added as a built in feature? It would be a great contest to support as many folks are learning cw this way and getting into contesting.

Details of the CWT are here: http://cwops.org/cwt.html

At the least I would love a pointer on how to multiply the score by the total number of unique contacts per band?

Thanks in advance, no worries if not...

John

dl1jbe, 2017-02-06:
Hi John,

you're request comes at the right time. Ervin HA2OS was working lately on minitest like contests which also requires to count unique calls as multis.

So please try the master branch from github and try to set UNIQUE_CALL_MULTI=ALL as well as ONE_POINT in the rule file.

Be aware that it is really fresh code and maybe there are some glitches. Feedback is welcome.

73, Tom DL1JBE

dl1jbe, 2017-03-18:
Seems to be solved. So I am closing it.
Feel free to reopen if needed.

closed 2017-03-18 by dl1jbe