Tlf/tlf #79: Documentation disagrees with behavior on behavior of Enter key in S&P mode

tnalpgge, 2017-12-04:
Recently I acquired a very inexpensive USB serial CW interface.  I was trying to debug why **cwdaemon** turns my call sign from `W6` into `M6`, and `CQ` into `KQ` (i.e. drops the first dah or dit).  As a workaround I asked **tlf** via rules files to send some extra characters to **cwdaemon** to make it "wake up" and send complete characters, e.g.:
```
F6=+- %
```
The [manual page](/Tlf/tlf/blame/35320fcc86b73cf22ef967ba5ee7f1714c7e1902/tlf.1.in#L532) claims that *Enter* sends the `MY` message as if I had hit *F6*.  My ears tell me otherwise; if it were doing that my call would not have been clipped to sound like `M6`.

Other bits of documentation, had I encountered them earlier, would have given me the obvious solution:
```
S&P_CALL_MSG=+- %
```
But the description of what the *Enter* key does in S&P mode is still a bit misleading.

I plan on submitting a pull request with documentation changes to match the observed behavior of the code.

airween, 2017-12-04:
Which cwdaemon are you using? As I remember, there is a buggy version, which has this or similar bug.
73, Ervin

tnalpgge, 2017-12-05:
Using **cwdaemon** version 0.10.1.

dl1jbe, 2017-12-05:
Hi,

Am Sun, 03 Dec 2017 20:10:45 -0800
schrieb tnalpgge <notifications@github.com>:

> Recently I acquired a very inexpensive USB serial CW interface.  I
> was trying to debug why **cwdaemon** turns my call sign from `W6`
> into `M6`, and `CQ` into `KQ` (i.e. drops the first dah or dit).  

That sounds as if keying start before your TX goes up fully. Have a
look at cwdaemons settings for PTT delay. You can configure it so that
keying starts later.

> As
> a workaround I asked **tlf** via rules files to send some extra
> characters to **cwdaemon** to make it "wake up" and send complete
> characters, e.g.: ``` F6=+- % ``` The [manual [...]
I just reread the chapter in the man page and they are misleading (and
partly wrong).
> 
> I plan on submitting a pull request with documentation changes to
> match the observed behavior of the code.
> 
Please do. Any contribution is welcome.

73, de Tom DL1JBE

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

PR #80 (2017-12-06): Correct documentation on Enter key to match observed behavior
Fix #79
closed 2017-12-06 by dl1jbe