Tlf/tlf #137: Page UP / Page DOWN to change reports

M0ORY, 2019-11-03:
This should really be configurable as not all keyboards these days have PgUp and PgDown keys. Macs in particular as the OS has no concept of PgUp / Down and so even plugging in an IBM keyboard these keys are not interpreted as intended meaning it's impossible to change signal reports on a mac either via ssh through the terminal or via a virtual machine console. 
Ideally I would use the up down arrows to change reports or just have the option to re-key them as a normal exchange field. 

I've been looking at the code on and off for the last year trying to figure this out for myself however it seems to be slightly beyond my skill level.

Cheers
Ben

zcsahok, 2019-11-04:
Up arrow would clash with log editing. We can check however if shift-up/down could be an option. The keys are currently hardcoded.
Thanks for the report.

zcsahok, 2019-11-04:
BTW did your try `Fn`+`Up`/`Down`?

https://apple.stackexchange.com/questions/26930/how-can-i-achieve-page-up-and-page-down-in-os-x

M0ORY, 2019-11-05:
Yes I've tried that in multiple ways. Via SSH from the Mac Terminal, via Virtual Box into an Xubuntu terminal and via VirtualBox into a console. None appear to work however when reading through the source last night I noticed that in CW mode pgup pgdn also changes the CW keyer speed and this does appear to be changing suggesting that the key combo does work. To test this I also tried it in Vim to see if it worked at all and Vim does interpret that key combination as pgup pgdn from a functional perspective .

Also looking at line 446 and down of callinput.c it looks like this is supposed to change CW speed if there is nothing in the callsign field but if the callsign is populated it should change the RST provided the change_rst variable = 1. A little digging later and I see the man page states that CHANGE_RST should be set (doesn't say where or how) I tried putting CHANGE_RST in logcfg.dat and now it works through the mac terminal but not through a VBox console or Xubuntu terminal within virtual box. 

It would probably be useful to have this (and any other such options) as a commented out option in the default logcfg.dat as it would be much more obvious to the user especially those of us with sponky keyboards requiring work arounds which may or may not work given certain scenarios. There's also nothing in the help file or the old documentation stating that this needs to be set in order for it to work. 

I'd still prefer to be able to tab across and key in a number as even with pgup pgdn working I can only change the 2nd digit and only of the RST I'm sending. If I'm given a 448 report on a contest the only way I change it from the default is by manually editing the log line with the :edit command after the QSO has already been incorrectly logged. I can deal with not being able to change the report I'm sending and just always send 599 but when someone throws a curve ball signal report at you mid contest you have to add a comment to change it later or just go edit the log  line. I know it almost never happens on contests but it is more common on general QSOs and while there is less of a need to keep the pace in that regard it's still annoying that you can't log a signal report as required without resorting to a text editor.

dl1jbe, 2019-11-05:
> I'd still prefer to be able to tab across and key in a number as even with pgup pgdn 
> working I can only change the 2nd digit and only of the RST I'm sending.

PgUp/PgDn does change the received report if you are in the exchange field.

But I agree at least in 'qso' mode it would be nice to be able to insert any given report.

M0ORY, 2019-11-05:
Ahh thanks that's good to know

zcsahok, 2019-11-06:
The key point is to find good a way to add this feature so that it's usable without disturbing the existing QSO input state machine.

My first though would be to define a new key _X_ that selects the corresponding (now inactive) RST field a bit like Tab. 

Not an easy one, for sure.

M0ORY, 2019-11-09:
Just for clarity It does work as intended but only when you add CHANGE_RST to the log config which isn't obvious from the available documentation.

dl1jbe, 2019-11-10:
>  It does work as intended but only when you add CHANGE_RST to the log config which isn't obvious from the available documentation.

It is documented in the man page (see PgUp or PgDn there).

M0ORY, 2019-11-10:
That section of the man page makes no mention of where to set CHANGE_RST and its hiding way down the page. This would be much more obvious if it were included in the default log config even if commented out.

dl1jbe, 2019-11-14:
So why not prepare a patch for logcfg.dat showing how you would present it and  open a pull request for it. Just go ahead.

dl1jbe, 2020-05-12:
Solved via PR #172. Closing.

closed 2020-05-12 by dl1jbe