Tlf/tlf #138: QSO counts per band are not shown when opening a non-empty log

zcsahok, 2019-11-06:
Log is loaded but QSO counts are 0.

[image]

Should be (and was) like this:

[image]

Caused by a change in 766e0f4.
The root cause is that functions are not (only) doing what their name says.
Original code (main.c):

```
scroll_log();
nr_qsos = readcalls();
```
Looks not too logical: why to call `scroll_log` before actually loading the log?

Updated code:
```
nr_qsos = readcalls();
scroll_log();
```

But at the end of `readcalls` the counts (and other stuff) are reset if `qsonum==1`.
And `qsonum` is set not in `readcalls` as one would expect, but in `scroll_log`...
So if they are called in "wrong" order the QSO counts are lost. (and maybe other info also, who knows)

dl1jbe, 2019-11-07:
Confirmed. Will be fixed soon. 

The check at end of readcalls is a leftover from long ago, but it seems it has to be dropped now.

dl1jbe, 2019-11-08:
Both problems are fixed. Please test and close the issue if no more problems left

zcsahok, 2019-11-08:
Both are fixed, closing the issue. Thanks.

closed 2019-11-08 by zcsahok