Tlf/tlf #144: Return from terminal editor results in arrow keys not working

N0NB, 2019-12-02:
I noticed when testing some fixes for CT mode that using a terminal mode editor results in the arrow keys only putting their letter representations--A, B, C, or D--into an entry field.

To reproduce, use the :SET, :CFG, or :EDIT command with a terminal editor defined in logcfg.dat.  When the editor is exited and Tlf resumes control of the screen, try using the keyboard arrow keys.

I have confirmed this with vim, nano, and mcedit.  My one try with gedit, a GUI editor, that did not take control of the terminal screen resulted in normal arrow key behavior when gedit was exited and control was resumed by Tlf.

I suspect we're not resetting or configuring the terminal completely/correctly after the terminal editor exits.

dl1jbe, 2019-12-02:
Hi,

Thanks for the report.
I can confirm it here. Let me look into it.

Am Mon, 02 Dec 2019 10:53:21 -0800
schrieb Nate Bargmann <notifications@github.com>:

> I noticed when testing some fixes for CT mode that using a terminal
> mode editor results in the arrow keys only putting their letter
> representations--A, B, C, or D--into an entry field.
>
> To [...]

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

zcsahok, 2019-12-02:
`logview` (less) works. It seems adding `endwin()` before and `clear_display+refreshp` after `system` helps.

N0NB, 2019-12-02:
You're correct, Zoli, the log view does work here as well.  I failed to note that originally.

zcsahok, 2019-12-02:
This worked for me:
```
    endwin();        //<<<

    cmdstr = g_strdup_printf("%s %s", editor, filename);
    retval = (system(cmdstr));;
    if (WEXITSTATUS(retval) == 127) {
	TLF_LOG_WARN("Can not start editor, check EDITOR= command");
    }
    g_free(cmdstr);

    clear_display(); //<<<
    refreshp();      //<<<
```

dl1jbe, 2019-12-02:
Right. with some corrections.

> This worked for me:
> 
> ```
>     endwin();        //<<< That one is needed
> 
>     cmdstr = g_strdup_printf("%s %s", editor, filename);
>     retval = (system(cmdstr));;

>     refreshp();      //<<< Has to be here so that we can display the TLF_WARN 

>     if (WEXITSTATUS(retval) == 127) {
> 	TLF_LOG_WARN("Can not start editor, check EDITOR= command");
>     }
>     g_free(cmdstr);
> 
>     clear_display(); //<<< That one can be left out
> ```

dl1jbe, 2019-12-02:
Fixed version in repository. Please test and close issue if fixed.

N0NB, 2019-12-03:
I'll test this, Tom.  BTW, I have the warnings generated by parse_logcfg.c fixed in a branch I'll be pushing out and issuing a pull request from in a day or two.

N0NB, 2019-12-03:
My quick test shows this is fixed.  Thanks, guys!

closed 2019-12-03 by N0NB