Tlf/tlf #148: Unrecognized configure option --enable-hamlib

jo3sn0w, 2019-12-15:
Page visited:
https://tlf.github.io/

Pulled source archive from download link:
http://download.savannah.gnu.org/releases/tlf/tlf-1.4.0.tar.gz

OS:  Debian 10 Buster
DE:  LXQt
Kernel:  4.19.0-6-amd64

----------

When I try the following:

`./configure --enable-hamlib --enable-fldigi-xmlrpc`

I am receiving below error at the beginning and end of configure script:

`configure: WARNING: unrecognized options: --enable-hamlib`

Somewhere in the middle I also see:

`configure: WARNING: you will need cmocka for running unit tests`

Page visited:
https://tlf.github.io/

Pulled source archive from download link:
http://download.savannah.gnu.org/releases/tlf/tlf-1.4.0.tar.gz

Hamlib is installed on my system:

`hamlib: /usr/include/hamlib /usr/share/man/man7/hamlib.7.gz`

I installed the suggested dependencies and header files including optional sox and xplanet.  Unsure what I'm missing other than a possible reboot.  If interested, the full output can be found here:

https://pastebin.com/vVecDeDe

Thank You

zcsahok, 2019-12-15:
As Hamlib is now mandatory the option to enable it has been removed.
It's found in your case (line 223) so all looks fine

```
checking for hamlib >= 1.2.8... yes
```

cmocka is only needed for tests. Just go ahead with compilation.

(I also use Buster and have the packaged tlf installed to be able to use the data files under /usr/share. For build configuration I use
```
./configure --prefix=/usr --datadir=/usr/share --enable-fldigi-xmlrpc
``` 
)

dl1jbe, 2019-12-15:
Fixed install instructions on 

https://tlf.github.io/

Thanks for pointing it out.

closed 2019-12-15 by dl1jbe

jo3sn0w, 2019-12-15:
Wow!  Appreciate the quick reply  :+1: