Tlf/tlf #150: Unclear about autoreconf --install

jo3sn0w, 2019-12-15:
Page visited:
https://tlf.github.io/

Pulled source archive from download link:
http://download.savannah.gnu.org/releases/tlf/tlf-1.4.0.tar.gz

OS: Debian 10 Buster
DE: LXQt
Kernel: 4.19.0-6-amd64

From within unpacked directory `tlf-1.4.0/`, I am receiving four of the same error:
`Can't open perl script "generate_defs.pl": No such file or directory`

Should I be running `autoreconf --install `after `make install` process instead?  Appreciate your help on my previous inquiry.

Thank You

dl1jbe, 2019-12-16:
That perl script is only needed by `autoreconf -i` to prepare the package for public release and not needed afterwards. 

So you should not see that message at all. 

Installation is a simple sequence of

`./configure`
`make`
`make install`

During which step you get the message about the missing perl script?

zcsahok, 2019-12-16:
@jo3sn0w: `autoreconf --install` is not needed when building from a release. Just use the 3 steps as indicated above.

@dl1jbe: `test/generate_defs.pl` is missing from the dist tar.gz. I could reproduce the issue and it comes from autoreconf.

N0NB, 2019-12-16:
I have long been a bit unclear whether autoreconf should work from a distributed source archive.  In theory, it should, at least according to hints in the Autoconf manual.  However, I suspect that is a remnant from the days when access to the SCM repository wasn't always possible.  These days it is easy to point someone toward a Git repository and ask them to work from there if they need to do something that requires autoreconf.

dl1jbe, 2019-12-16:
The only reason to run `autoreconf` I see for distribution package managers who needs to adapt `configure.ac`.

But anyway I will add the file to the next release. Until then we should keep the issue open.

closed 2019-12-23 by dl1jbe