Tlf/tlf #163: ALT-T for tune does not work anymore

dh5fs, 2020-03-15:
tested with latest build from master, 20200315
laste working version here 1.3.0
cwdaemon version 0.10.2

dl1jbe, 2020-03-15:
Works here with tlf-1.32., tlf-1.4 and master branch with cwdaemon and winkeydaemon

Try to start cwdaemon with the following parameters:
'-n' (stay in foreground) and '-y d' (enable debug output).
Please check if it receives commands from tlf and report back.

zcsahok, 2020-03-20:
No issue with cwdaemon 0.10.2-2 and tlf built from master on 19.3.

airween, 2020-03-20:
```
$ cwdaemon -V
cwdaemon version 0.10.1
$ tlf -V
tlf-1.5~git
$ git rev-parse HEAD
00b5d6ba157d55763cb98bca23b1592992016c07
```
(This is the current master on 20th of March.)

Works as well.

dl1jbe, 2020-03-29:
@dh5fs Any news Fred?

dl1jbe, 2020-05-23:
Closing it, as we can not confirm it and further information are missing.
Please reopen if needed.

closed 2020-05-23 by dl1jbe