Tlf/tlf #166: Hamlib not found, but is installed

classabbyamp, 2020-03-30:
I downloaded the latest source archive from savannah.gnu.org, and am attempting to package `tlf` for void linux.

`autoreconf -i` runs fine.

During `./configure --prefix=/usr --sysconfdir=/etc --infodir=/usr/share/info --mandir=/usr/share/man --localstatedir=/var --datadir=/usr/share --enable-fldigi-xmlrpc` (all options before `--datadir` are set by void), I get this error:

```
checking for hamlib >= 1.2.8... no
configure: error: Hamradio control libraries 1.2.8 or later not found...
```

The `hamlib` install:
- The current version of `hamlib` in the void repos is 3.3.
- The `hamlib` directory exists in `/usr/include`
- `hamlib.pc` exists in `/usr/lib/pkgconfig`
- `libhamlib{,++}.so{,.2,.2.1.3}` exist in `/usr/lib`

Any suggestions?

classabbyamp, 2020-03-30:
https://github.com/Hamlib/Hamlib/wiki/FAQ#i-write-applications-what-do-i-need-to-do-to-link-to-hamlib

From what I see in configure.ac, it is not how hamlib says to link to it.

airween, 2020-03-30:
Hi @classabbyamp,

I'm trying to reproduce your steps:

```
wget http://download.savannah.gnu.org/releases/tlf/tlf-1.4.0.tar.gz
...
tar -xzf tlf-1.4.0.tar.gz
cd tlf-1.4.0/
$ autoreconf -i
Can't open perl script "generate_defs.pl": No such file or directory
Can't open perl script "generate_defs.pl": No such file or directory
Can't open perl script "generate_defs.pl": No such file or directory
Can't open perl script "generate_defs.pl": No such file or directory
$ ./configure --prefix=/usr --sysconfdir=/etc --infodir=/usr/share/info --mandir=/usr/share/man --localstatedir=/var --datadir=/usr/share --enable-fldigi-xmlrpc
...
checking for HAMLIB... yes
checking for rig_open in -lhamlib... yes
checking hamlib/rig.h usability... yes
checking hamlib/rig.h presence... yes
checking for hamlib/rig.h... yes
...
```

Could you show us your `config.log`? I mean something like this:
```
configure:7276: checking for HAMLIB
configure:7283: $PKG_CONFIG --exists --print-errors "$hamlib_modules"
configure:7286: $? = 0
configure:7300: $PKG_CONFIG --exists --print-errors "$hamlib_modules"
configure:7303: $? = 0
configure:7341: result: yes
configure:7349: checking for rig_open in -lhamlib
configure:7374: gcc -o conftest -g -O2   conftest.c -lhamlib   -L/usr/local/lib -lhamlib >&5
configure:7374: $? = 0
configure:7383: result: yes
configure:7405: checking hamlib/rig.h usability
configure:7405: gcc -c -g -O2 -pthread -I/usr/local/include -I/usr/include/libusb-1.0  -I/usr/local/include -I/usr/include/libusb-1.0 conftest.c >&5
configure:7405: $? = 0
configure:7405: result: yes
configure:7405: checking hamlib/rig.h presence
configure:7405: gcc -E  -I/usr/local/include -I/usr/include/libusb-1.0 conftest.c
configure:7405: $? = 0
configure:7405: result: yes
```

What's at your side?

classabbyamp, 2020-03-30:
here's config.log: https://gist.github.com/classabbyamp/0bd47920e8b0dd76f77e103e887a6a65

also, as a note, I'm doing this within void's xbps package manager, so by default, it will do `./configure; make; make install` with arguments the you provide.

classabbyamp, 2020-03-30:
ok that's interesting:

```
configure:7403: checking for hamlib >= 1.2.8
configure:7410: $PKG_CONFIG --exists --print-errors "$hamlib_modules"
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libusb-1.0', required by 'hamlib', not found
configure:7413: $? = 1
configure:7427: $PKG_CONFIG --exists --print-errors "$hamlib_modules"
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libusb-1.0', required by 'hamlib', not found
configure:7430: $? = 1
configure:7444: result: no
Package 'libusb-1.0', required by 'hamlib', not found
configure:7460: error: Hamradio control libraries 1.2.8 or later not found...
```

airween, 2020-03-30:
Looks like the [`libusb`](https://github.com/libusb/libusb) is missing from your system.

classabbyamp, 2020-03-30:
Yep, that was it. Thanks!

closed 2020-03-30 by classabbyamp