Tlf/tlf #180: Kansas QSO Party mult and dupe issues

N0NB, 2020-08-29:
I will be adding comments to this issue through the weekend.

1) I changed bands from 20 CW to 40 CW early on this morning.  I had at least three QSOs that I had worked on 20 that were then worked on 40.  Of those QSOs it appeared that each had been the first mult on 20 and they were then listed as mults on 40.

2) I changed from CW to phone and now each station that I worked on CW is now showing as a dupe on phone.

After the event is over tomorrow at 2000z I can do a more thorough investigation.

N0NB, 2020-08-29:
QSY to 20 phone and the first station I worked was new but the mult was counted as new even though it had already been logged as a new mult on 40 CW.

N0NB, 2020-08-29:
```
#################################
#      Scoring rules            #
#################################
#
MIXED
RECALL_MULTS
SSBPOINTS=2
CWPOINTS=3
SECTION_MULT
### WYSIWYG_ONCE
MULT_LIST=ksqp.txt
####### END #####################
```

Relevant rules section.

N0NB, 2020-08-29:
BTW, have compiled from the master branch as of 26 August.

N0NB, 2020-08-29:
Just worked one of the mobile stations (W0B, W0BH our QSO party coordinator) also in Marshall County (MSH) and the section scoring worked as advertised counting the QSO for the KS mult.

N0NB, 2020-08-30:
For comparison, here is what Tlf is scoring for yesterday:

```
 Band   160   80   40   20   15   10
 QSO's    0  113  180  159    0    0
 Mult     0   33   40   36    0    0
                                                                                
Pts: 1237  Mul: 109 Score: 134833 
```

Processing through a custom Perl script I wrote a few years ago is more realistic:

```
$ ./log-calc.pl

Number of mults: 48

Mults/QSOs:
AL:   5  AR:   6  AZ:   3  BC:   1  CA:  36  CO:   3  CT:   7  DE:   1  
DX:   3  FL:  16  GA:  14  IA:   4  IL:  16  IN:  11  KS:   6  KY:   6  
LA:   2  MA:   7  MD:   4  ME:   4  MI:  11  MN:  32  MO:  12  MS:   6  
NC:   9  NE:   3  NH:   5  NJ:  12  NM:   5  NV:   1  NY:  14  OH:  32  
OK:   7  ON:   6  OR:  10  PA:  25  RI:   1  SC:   6  SD:   2  SK:   1  
TN:  13  TX:  22  VA:  18  VT:   4  WA:  14  WI:  11  WV:  12  WY:   3  

80m CW: 100       SSB: 11        Digital: 0         
40m CW: 155       SSB: 21        Digital: 0         
20m CW: 142       SSB: 13        Digital: 0         
15m CW: 0         SSB: 0         Digital: 0         
10m CW: 0         SSB: 0         Digital: 0         

Totals
All: 452  Duped: 442  
CW: 397  SSB: 45   Digital: 0    

Points:
Pts CW: 90          SSB: 1191        Digital: 0           Total: 1281

Total claimed score: 61488
```

dl1jbe, 2020-09-01:
Hi Nate,

just coming back from holiday vacation I saw your mails.
Can you please post a link to the contest rules?

73, de Tom DL1JBE

Am Sun, 30 Aug 2020 06:53:07 -0700
schrieb Nate Bargmann <notifications@github.com>:

> For comparison, here is what Tlf is scoring for yesterday:
>
> ```
>  Band   160   80   40   20   15   10
>  QSO's    0  113  180  159    0    0
>  Mult     0   33   40   36    0 [...]

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

airween, 2020-09-01:
> Can you please post a link to the contest rules?

looks like you answered by e-mail - just fyi: he already [did](https://github.com/Tlf/tlf/issues/180#issuecomment-683313710) it.

N0NB, 2020-09-01:
Hi Tom.

The rules are here: https://www.ksqsoparty.org/rules/KSQPRules2020.pdf?1

I was operating as a single-op, low power, in state.

For in-state ops there are a total of 64 multipliers--50 US states, 13 Canadian provinces, and DX for all other areas.  Mults count once.  In-state ops exchange county designators and the first such exchange counts for KS.  This did seem to work except for the general error of counting a mult as new on each band it was first worked.

Stations are counted for points once per band/mode.

If it helps I can email you my log, rules, and logcfg.dat files privately.

N0NB, 2020-09-01:
Here is the final comparison of what Tlf scored and my custom script:

```
Band   160   80   40   20   15   10
QSO's    0  113  284  286    0    0
Mult     0   33   42   39    0    0

Pts: 1745  Mul: 114 Score: 198930 
```

And my script which reads the Cabrillo file:

```
$ ./log-calc.pl

Number of mults: 52

Mults/QSOs:
AK:   1  AL:   6  AR:  10  AZ:   7  BC:   3  CA:  61  CO:  13  CT:  10  
DE:   2  DX:   6  FL:  21  GA:  17  IA:   5  ID:   1  IL:  28  IN:  18  
KS:   6  KY:   8  LA:   2  MA:  10  MD:   5  ME:   6  MI:  15  MN:  39  
MO:  17  MS:   8  NB:   1  NC:  16  NE:   3  NH:   7  NJ:  19  NM:   8  
NV:   3  NY:  23  OH:  44  OK:   8  ON:  13  OR:  16  PA:  39  QC:   2  
RI:   2  SC:  12  SD:   2  SK:   1  TN:  19  TX:  40  VA:  25  VT:   4  
WA:  17  WI:  15  WV:  15  WY:   4  

80m CW: 100       SSB: 11        Digital: 0         
40m CW: 199       SSB: 72        Digital: 0         
20m CW: 224       SSB: 52        Digital: 0         
15m CW: 0         SSB: 0         Digital: 0         
10m CW: 0         SSB: 0         Digital: 0         

Totals
All: 683  Duped: 658  
CW: 523  SSB: 135  Digital: 0    

Points:
Pts CW: 1569        SSB: 270         Digital: 0           Total: 1839

Total claimed score: 95628
```

This is the best result in my 11 years of operating this event and it looks like I'll get second place, at best, in my category.  :-/

dl1jbe, 2020-09-02:
> Hi Tom.
> 
> The rules are here: https://www.ksqsoparty.org/rules/KSQPRules2020.pdf?1
> 
Thanks. I had a look at it. I remember we were discussing it already in 2017.

> 
> Stations are counted for points once per band/mode.
> 
- Here is problem 1:
SECTION_MULT does count mults per band. While we can add a special keyword SECTION_MULT_ONCE for the problem it is no good long time solution.
Here a more general approach to the mult scoring is needed in long term.

> If it helps I can email you my log, rules, and logcfg.dat files privately.

That would be helpful, please do. I can then check my work on the code with your results.

- Second problem seems that SSB QSOs are displayed as dupe in CW and vice versa. That should not happen in MIXED_MODE.

- Third problem seem to be the different QSO counts from tlf and your perl script. E.G. TLF counts 113 QSOs on 80 and your script shows only 111. Similar on other bands.

And by the way: congrats to the good results.

N0NB, 2020-09-12:
Tom, I remembered to send you the log and several files in a private email.

dl1jbe, 2020-09-19:
Nate, the problem should be fixed with commit  92003563245140ab200a56f49818371eaeca5bf0.

Please build that version and try the following:

1. export your existing log to cabrillo (and make a copy of the log).
2. Change SECTION_MULT keyword to SECTION_MULT_ONCE (counts multis only once).
3. Reimport the cabrillo log via 'tlf -i'
4. cp IMPORT_ksqp_2020.log ksqp_2020.log
5. Start tlf  (should give the same results as your script now).

dl1jbe, 2020-10-24:
Closing. Please reopen if problem still exists.

closed 2020-10-24 by dl1jbe