Tlf/tlf #182: Longer multiplier ignored in favor of shorter

tnalpgge, 2020-10-03:
I am preparing for the [California QSO Party](https://cqp.org/).  As an in-state participant, I have to record not only California counties but also American states and Canadian provinces as [multipliers](http://cqp.org/cqp_multipliers.html).

While testing out my setup it appears that the multiplier for Los Angeles County (`LANG`) is treated as the state of Louisiana (`LA`).  Perhaps it is not looking for the longest match when it should be?

Here is where I simulated the log entry and used `\` to store it.

[image]

And here is how it looks in the `:mult` screen.

[image]

I am using a build from the Git sources at this commit:
```
: [tony@medora tlf]$ ; git log -n 1
commit 7f95002f66beaa1cf1af64ffc0c4333858bc0da1 (HEAD -> master, kcjcw)
Author: Thomas Beierlein <tomjbe@gentoo.org>
Date:   Fri May 1 10:29:54 2020 +0200

    fix compiler warnings
```
...installed into `/usr/local`. 

About my machine:

```
: [tony@medora tlf]$ ; cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.5 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.5 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
: [tony@medora tlf]$ ; uname -a
Linux medora 4.15.0-111-generic #112-Ubuntu SMP Thu Jul 9 20:32:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

tnalpgge, 2020-10-03:
My `logcfg.dat`:

```
EDITOR=vi
CALL=W6SFI
CWSPEED=20
CLUSTER
CHECKWINDOW
PARTIALS
RIGMODEL=2
RIGSPEED=9600
RIGPORT=localhost:4532
RADIO_CONTROL
SIDETONE_VOLUME=0
BANDMAP=BMD,300
NETKEYER
RULES=california-qso-party
```

The top of `california-qso-party`:

```
########################
# California QSO Party #
########################
CONTEST=california-qso-party
LOGFILE=california-qso-party.log
CABRILLO=UNIVERSAL
CWPOINTS=3
SSBPOINTS=2
SERIAL+SECTION
MULT_LIST=/home/tony/.config/tlf/multipliers/california-qso-party-in-state
```
(keyer messages not displayed)

An excerpt from the multipliers file `california-qso-party-in-state`:

```
: [tony@medora tlf]$ ; grep ^L multipliers/california-qso-party-in-state
LAKE 
LASS 
LANG 
LA 
```

Predictably, Lake (`LAKE`) and Lassen (`LASS`) Counties also are treated as if they were Louisiana.

tnalpgge, 2020-10-03:
A workaround is going back to edit the log with `:edi`, whereupon the multipliers will be honored.

dl1jbe, 2020-10-05:
Thanks for the report. We will look into it in next days.

dl1jbe, 2020-10-15:
Problem found. TLF does only recognize section multipliers with maximal 3 characters. Seems to be a leftover from times where 3 letter sections were the only ones.
I will extend the recognized length to (6 or 8 characters). Lets hope that no 'creative' author of contest rules will introduce longer sections.

dl1jbe, 2020-10-16:
Just reread your message and the contest rules, it seems to me that as an in-state amateur your californian sections do NOT count as separate multiplier but only once for CA. Bt you have to record them in the log.

If that interpretation is correct please have a look for the MULT_LIST entry in the man page and especially for the use of aliases. That would mean you have to write something like

CA:LANG,LASS,LAKE 

in your mult file.

Anyway the 3 letter section limit needs still to be resolved.

dl1jbe, 2020-10-24:
Fixed in PR #183. 

Thanks again for reporting.

closed 2020-10-24 by dl1jbe