Tlf/tlf #185: Can't change interface colors

kostecke, 2020-10-26:
[image]

I'm trying to use tlf in an xterm (under the Xmonad window manager) and am unable to configure readable colors. The window shot linked at the beginning of this comment shows what I'm stuck with.

The TLFCOLOR settings listed in the man page have no effect.

And there's practically no helpful information available on-line about how to set colors. What am I supposed to do?

dl1jbe, 2020-10-26:
TLFCOLOR does work only in plain linux text console and not for X window terminals.
For xterm you can use a .Xressources file in your home directory. From version 1.4.1 TLF has a sample file in its doc directory (thanks to N0NB for that).
If you have an older version just look into the doc directory here on github for the file.

kostecke, 2020-10-26:
From what I can see in the sample [.Xresources file](https://github.com/Tlf/tlf/blob/master/doc/Xresources) I will have to change my ideal global terminal color scheme to support software I will likely use only infrequently.

This is rather unfriendly.

Users should be able to specify tlf specific colors (e.g. tlf*color0: ).

dl1jbe, 2020-10-26:
That would be good. But as far as I see that feature is only supported for native X based applications. As TLF is curses based I am not sure that it will respret such settings. 
But we can try.

I you change your TERM settings away from 'xterm' or 'rxvt' the TLFCOLORS= settings will be respected. You can try it as a second way.

kostecke, 2020-10-27:
>I you change your TERM settings away from 'xterm' or 'rxvt' the TLFCOLORS= settings will be respected. You can try it as a second way.

Thank-you for that suggestion; it's a good one.

After posting my previous comment I remembered the `xterm -class ...` workaround (I first used it back in 1998) which can be used to implement fine grained overrides of general Xterm settings (e.g. colors).

Thanks for the discussion!

closed 2020-10-27 by kostecke

dl1jbe, 2020-10-27:
Very well. Did not know about -class trick and have still trouble to get it working. But I managed to use -name 

in .Xresource have the colors defined for application name 'tlf'

   tlf*color0:  #........

and start the program with

   xterm -name tlf -e tlf <parameters>

Thanks for the hint.

zcsahok, 2020-10-27:
my 2 cents: I use tlf in an xfce4-terminal with `TERM=xterm` setting and quite happy with the default colors.

[image]

(here it runs in KDE environment)

dl1jbe, 2020-10-27:
Yep, that works very well for me too. Only downside is that xfce catches some of the special keys in default configuration. But luckily you can configure the terminal to pass through most of the keys.