Tlf/tlf #188: Exchange sent not populated in Cabrillo file

M0ORY, 2020-10-27:
I took part in CQWW ssb over the weekend and when I came to uploading my log I noticed the sent exchange sent field of the generated cabrillo file wasn't populated with my CQ zone and I had to edit the file manually and add it. 
 
In logcfg.dat I had "RULES=cqww" which seems to set things up for CQWW and indeed auto populates the received exchange field with the cq zone based on what we expect from the callsign. 

I see in the rules file for CQWW the cabrillo format is set to CQWW and the cabrillo.fmt entry contains the exchange sent (EXC_S) field in the format however I can't see any documentation showing how the operators region can be set so I assume it's supposed to work it out from "CALL=callsign" 

This was tested on my raspbian instance installed from apt running v1.2.4.1 so I've installed the latest v1.4.1 from source and I see from the startup screen when starting with the -v flag it does indeed seem to assume my zone and continent however still no exchange sent is populated in the cabrillo file. Even if it was ignoring the cqww format it should be using the universal format which also includes the exchange sent. I altered the format and added multiple instances of the mode field and re-created the cabrillo file and that change was evident in the file so it is using the correct format.

I've also tried configuring a general contest using the universal cabrillo format and with SERIAL_EXCHANGE which should populate each line with the incremental serial number however that doesn't seem to appear in the cabrillo file either

It would make sense to be able to configure this sort of thing from logcfg.dat as the assumed region isn't always accurate as I noticed some of the north american stations gave different zones to what was assumed from their callsign and this would also be the case if operating maritime mobile from a different zone to the country that issued the call.

dl1jbe, 2020-10-28:
When you start writing the cabrillo file with ':write' the first thing you get asked is the exchange Tlf should put into the file. I fear you skipped to insert your CQ zone in that step?

M0ORY, 2020-10-28:
Thanks, You are indeed spot on. This is definitely a PICNIC case (Problem In Chair Not In Computer) 

I have always skipped the options there because I often find that the headers that most of the questions populate often aren't relevant to the contest in question and I just copied that from the contests published cabrillo format and filled it out manually. This is because I mostly use TLF for unsupported contests such as UKAC VHF / UHF so I have to mangle the cabrillo file anyway because those bands aren't supported. I do this because I use a mac and there's very little in the way of logging software on mac. Running loggers in VM's can be problematic because when the mac sleeps the VM pauses which also pauses it's clock leading to all further log lines having skewed timestamps so I run TLF on a Raspberry PI and SSH to it.

Thanks again for your help
Ben M0ORY

closed 2020-10-28 by M0ORY