Tlf/tlf #198: Add a way of specifying Cabrillo header values

zcsahok, 2020-12-15:
Currently Cabrillo header values (CONTEST, CATEGORY-*, ...) have to be entered online while exporting the log. This process is quite tedious and error prone.

As a user I expect a way to specify such mostly seldom changing infos both
* via config file (logcfg.dat / rule file)
* and by reusing previous or sample logs as a template.

The procesing of configuration shall fit into the current Cabrillo QSO format specification logic (keyword CABRILLO).

airween, 2020-12-15:
To anybody: a possible solution was introduced [here](https://github.com/Tlf/tlf/pull/196#issuecomment-745111450).

zcsahok, 2020-12-15:
It would be nice to have a way of setting mostly constant (or default) data in logcfg+rule, i.e. name, email, power, etc as
```
CABRILLO-NAME: Joe Doe
CABRILLO-EMAIL: j.doe@...
CABRILLO-CATEGORY-POWER: LOW
```

`CABRILLO-TEMPLATE` could point to an actual Cabrillo file from which all relevant fields  (excluding QSO, START/END, CREATED) could be taken over. Ideally a valid file should be usable as-is in order to be user friendly: just point to an example log and that's it.

Missing fields would be still asked for during Cabrillo creation.

`CABRILLO-TEMPLATE` would be logically part of rules. But even without a template one could describe the fields using `CABRILLO-*` configs.

The current `CABRILLO` QSO format spec could be also optionally merged to the template by adding the same kind of "QSO=" line manually. This would result in a compact spec at the expense of possibly duplicating formats per contest. (or QSO= could be even a reference to the format...)

E.g.
```
START-OF-LOG: 3.0  <- not used
CREATED-BY: tlf-1.3.0 <- not used
CONTEST: 9ACW
#CATEGORY-BAND: 80M ... commented out
CATEGORY-MODE: CW
CATEGORY-OPERATOR: SINGLE
CATEGORY-POWER: LOW
QSO=<FMT spec>

```

zcsahok, 2020-12-23:
_Further brainstorming..._
Tags fall into these categories -
* **ask**, with optional hint text in parentheses (or brackets?)
	`CABRILLO-CATEGORY-POWER=`
	or
	`CABRILLO-CATEGORY-POWER= (LOW,HIGH)` or `[LOW, HIGH]`
        Hint text would be shown when asking for a value.

* **fixed**, never or seldom changing value
	`CABRILLO-NAME= Joe Doe`
        Leading/trailing whitespace is removed.
	
* **disabled**, not to be written into the exported file
	`CABRILLO-CATEGORY-OVERLAY= -`
        Value is a single dash.

* **internal**, values used by Tlf internally like exchange value or QSO format
       `CABRILLO-EXCHANGE= #`
       `CABRILLO-QSO-FORMAT= universal` 

All values could be specified via logcfg/rules or via `CABRILLO-TEMPLATE` file. Latest definition takes precedence.

For reference the Cabrillo specification: https://wwrof.org/cabrillo/cabrillo-specification-v3/

airween, 2020-12-23:
Hi Zoli,

* _**ask**, with optional hint text in parentheses (or brackets?)_

sorry, I don't understand this question

_All values could be specified via logcfg/rules or via CABRILLO-TEMPLATE file. Latest definition takes precedence._

With the new parser can we check the config option where does it came from?

zcsahok, 2020-12-23:
"ask": I was just thinking if the syntax "(Value1, Value2)" or "[Value1, Value2]" would be better. One could imagine one using parentheses which would mislead the logic.
E.g.
`CABRILLO-SOAPBOX= (only cw)`
would result in asking for soapbox text with hint "only cw".
Maybe it's less likely that one writes
`CABRILLO-SOAPBOX= [only cw]`
Not sure, though.

We don't know where a config option comes from. I'm not sure if that's needed.
We could parse CABRILLO-TEMPLATE on the fly, so a sequence
```
# logcfg
CABRILLO-NAME = A

# rules
CABRILLO-TEMPLATE = my.cab
CABRILLO-NAME = B

# my.cab
NAME: C
```
results in NAME=B.
Reversing the lines in rules would lead to NAME=C.
Does this make sense?

zcsahok, 2020-12-28:
Pse check out https://github.com/zcsahok/tlf/tree/cabrillo_improvement

dl1jbe, 2020-12-29:
> It would be nice to have a way of setting mostly constant (or default) data in logcfg+rule, i.e. name, email, power, etc as
> 
> ```
> SHORT_SERIAL
> EDITOR=vi
> 
> CABRILLO-NAME= Joe Doe
> CABRILLO-EMAIL= j.doe@...
> CABRILLO-CATEGORY-POWER= LOW
> ```
Down side is we have to parse it during read of logcfg.dat and/or rules. As CABRILLO keywords evolve over time we would either have to a) adapt the parser often or b) lookup for the CABRILLO- prefix and use what comes behind as entry into the cabrillo log file.

> Additionally `CABRILLO-TEMPLATE` could point to an actual Cabrillo file from which all relevant fields (excluding QSO, START/END, CREATED) could be taken over. Ideally a valid file should be usable as-is in order to be user friendly: just point to an example log and that's it.

With these variation we have only one additional keyword and can use whatever is in the file. Update to new cabrillo versions (or contest specific settings) can be done easily by changing the template file.

> Missing fields would be still asked for during Cabrillo creation.
>
> `CABRILLO-TEMPLATE` would be logically part of rules. But even without a template one could describe the fields using `CABRILLO-*` [...]
As long as there are rule files for most of the contests that is a good solution. Otherwise anyone who wants to roll its own contest has to look how to prepare such a format specification. I am not sure if that is the right way to go.

Just my 2 ct's

dl1jbe, 2020-12-29:
>       `CABRILLO-CATEGORY-POWER= (LOW,HIGH)` or `[LOW, HIGH]`
>       Hint text would be shown when asking for a value.
> 
Good idea.

> 
>     * **disabled**, not to be written into the exported file
>       `CABRILLO-CATEGORY-OVERLAY= -`
>       Value is a single dash.
> 
Does it make sense? If we do not want it in the cabrillo file at all just drop it from the template.

dl1jbe, 2020-12-29:
> "ask": I was just thinking if the syntax "(Value1, Value2)" or "[Value1, Value2]" would be better. One could imagine one using parentheses which would mislead the logic.

How about using '|' between options to make clear they are alternatives, e.g. [Value 1 | Value 2 | ... ]. 

> E.g.
> `CABRILLO-SOAPBOX= (only cw)`

It is common to request sentences to be in quotation marks. So that should be:

`CABRILLO-SOAPBOX= ("cw only")` 

> would result in asking for soapbox text with hint "only cw".

> We don't know where a config option comes from. I'm not sure if that's needed.
> We could parse CABRILLO-TEMPLATE on the fly, so a sequence
>
> ```
> # logcfg
> CABRILLO-NAME = A
>
> # rules
> CABRILLO-TEMPLATE = [...]

It also depends on the order to evaluate the lines. We can parse my.cab on the fly during read of rules file or afterwards. That opens a can of worms.
Would it not be better to discourage the mixed use of a template file and single item settings?

zcsahok, 2020-12-29:
My assumption is that Cabrillo format as such is quite stable, so we can define all existing fields as configurable.
Only the ones currently used are enabled by default. (see `cabrillo_utils.c`  in my implementation mentioned above)

Writing of Cabrillo file is not different than now: it goes through the list of all fields and either

- writes the previously (via config or template) set value
- ask for a value if field is enabled (optionally with a pre-set hint)
- skips the field if disabled.

Disabling a field makes sense if one wants to suppress a particular field. Current code ask a lot of stuff (a bit annoying), but say half of them are not relevant for my case.

The "Cabrillo template" is not meant as an actual template that is filled in, but rather an additional way of defining field values. It is completely optional to use either templates or configs. The same goal can be reached using any of them.

Parsing order:

1. logcfg.dat: configs in file order; if a template is given it parsed on the fly
2. rule file: same, configs in file order; if a template is given it parsed on the fly

One kind of "normal" usage would be
- set fixed values in logcfg (like name, email, assisted, mode,...)
- set contents-specific stuff in rules file (contest name, mode,...) 
    here one has the option to overwrite values from logcfg (e.g. mixed mode contest, ...)

dl1jbe, 2020-12-29:
> My assumption is that Cabrillo format as such is quite stable, so we can define all existing fields as configurable.

'Stable' is a question of time. In the years from 2010 it changed quite a bit. See also my comment in the code about the ADDRESS(2) field. Similar goes for the SOAPBOX. 

A second problem I see is that different contests seems to use slightly different cabrillo settings (especially for the head lines). Maybe that is because they are sometimes a little bit behind the actual specification.

> Only the ones currently used are enabled by default. (see `cabrillo_utils.c` in my implementation mentioned above)
> 
Just had a look. Quite a fine peace of code.

> 
> Disabling a field makes sense if one wants to suppress a particular field. Current code ask a lot of stuff (a bit annoying), but say half of them are not relevant for my case.
> 
Ok. I misinterpreted you code in one point. I thoughed that you put only the line from template or CABRILLO-* keyword into the log. So you could leave them just out. But you use your fixed cabrillo_fields list instead and asks for fields not initialized yet. Then it makes sense to disable some of them.

 
> Parsing order:
> 
>     1. logcfg.dat: configs in file order; if a template is given it parsed on the fly
>     2. rule file: same, configs in file order; if a template is given it parsed on the fly
> 
That should get a little bit documentation. Otherwise we will get a lot of questions.

zcsahok, 2020-12-29:
My code assumes that each field has a unique name (id). This is not the case for ADDRESS and SOAPBOX. For these with suffixes (2) and (3) two-two additional fields are foreseen. Personally I don't use ADDRESS and one or none lines of SOAPBOX. I would assume 3 lines should be enough for each. (There are also specific fields like ADDRESS-CITY, these I didn't add yet).
So one can a config
```
CABRILLO-ADDRESS= 123 Street
CABRILLO-ADDRESS(2)= Mycity, Myland
```
results in
```
ADDRESS: 123 Street
ADDRESS: Mycity, Myland
```

Regarding the template: another option would be to use the fields (except of course START/END, CREATED and QSO/QTC) from the template as-is. In this case we would either

- use the generation based on configs (with optional asking the user)
- or take the fields from the template and add QSO/QTC lines.

But not both. So by specifying a template we would simply ignore all configs (both that were already processed and the ones to be processed).

The drawback is that then the contest template has no way for a selection (like `POWER: (QRP, LOW)`), all is taken verbatim.

The advantage is than any new field can be supported. But I guess this is just minor one if Cabrillo v3 stays stable.
Also multiple ADDR/SOAPBOX field can be easily specified.

Option2: drop template support all together.

Option3: drop configs and keep template only.

From usability perspective I'm fine with the two level (global + contest) config without template. Of course others may have another preferences.

airween, 2020-12-29:
Zoli and me have discussed in private about his finished code.

Today I grabbed this branch, and tried the CABRILLO template with my latest WAE contest log (I've made a copy before the test).

Here is how I used:

```
$ pwd
/home/airween/.tlf/wae2020cw_test
```
this is my "base" config in the `$HOME/.tlf` directory. Perhaps these lines will never changed.

```
$ cat ../HA2OS.cbr 
CONTEST:-
CATEGORY-ASSISTED:-
CATEGORY-BAND:-
CATEGORY-MODE:-
CATEGORY-OPERATOR:SINGLE-OP
CATEGORY-POWER:-
CATEGORY-STATION:-
CATEGORY-TIME:-
CATEGORY-TRANSMITTER:-
CATEGORY-OVERLAY:-
CERTIFICATE:-
CLUB:-
LOCATION:-
GRID-LOCATOR:-
NAME:Ervin
EMAIL:airween@gmail.com
ADDRESS:
ADDRESS(2):
ADDRESS(3):
OPERATORS:
OFFTIME:
SOAPBOX:
SOAPBOX(2):
SOAPBOX(3):
```
Note: fields with values `-` means that I didn't set up the field **but** the logger can ask me. Fields with values '' (empty) means I don't want to set up those later.

```
$ grep CABRILLO rules/waedc 
CABRILLO=WAEDC
CABRILLO-TEMPLATE=/home/airween/.tlf/HA2OS.cbr
CABRILLO-EXCHANGE=#
CABRILLO-CONTEST=WAEDC
CABRILLO-CATEGORY-MODE=CW
CABRILLO-CATEGORY-POWER=LOW
CABRILLO-CATEGORY-ASSISTED=
CABRILLO-CATEGORY-BAND=ALL
CABRILLO-CATEGORY-OPERATOR=SINGLE-OP
CABRILLO-CATEGORY-STATION=
CABRILLO-CATEGORY-TIME=
CABRILLO-CATEGORY-TRANSMITTER=
CABRILLO-CATEGORY-OVERLAY=
CABRILLO-CLUB=
CABRILLO-LOCATION=
```

When I started the logger and type `:wri`, it didn't asked me anything, just wrote the file - here is the result:

```
$ head -n 19 HA2OS.cbr 
START-OF-LOG: 3.0
CREATED-BY: tlf-1.5~git
CONTEST: WAEDC
CALLSIGN: HA2OS
CATEGORY-ASSISTED: 
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-POWER: LOW
CLAIMED-SCORE: 92722
NAME: Ervin
EMAIL: airween@gmail.com
ADDRESS: 
ADDRESS: 
ADDRESS: 
OPERATORS: 
SOAPBOX: 
SOAPBOX: 
QSO:  7001 CW 2020-08-08 0002 HA2OS         599 0001   VY2ZM         599 7     
```

I think this is the most convenient config what we can use :).

TNX Zoli again!

dl1jbe, 2020-12-30:
> My code assumes that each field has a unique name (id). This is not the case for ADDRESS and SOAPBOX. For these with suffixes (2) and (3) two-two additional fields are foreseen. Personally I don't [...]

Seems the standard allows both variants, so ADRESS/ADRESS(2)/ADDRESS(3) may be sufficient.
I am not sure how tolerant the robots used to send your log to are. So let us try to use the scheme -/(2)/(3) schemata for now.

> So a config
> 
> ```
> CABRILLO-ADDRESS= 123 Street
> CABRILLO-ADDRESS(2)= Mycity, Myland
> ```
> 
> results in
> 
> ```
> ADDRESS: 123 Street
> ADDRESS: Mycity, Myland
> ```

Fine, I overlooked the part where you dropped the (...) part of the key.

> Regarding the template: another option would be to use the fields (except of course START/END, CREATED and QSO/QTC) from the template as-is. In this case we would either
>
>     * use the generation [...]

I think it is better to allow a mix of both sources. Constant information (e.g. operator name, club or similar) should stay in the logcfg.dat.

> The drawback is that then the contest template has no way for a selection (like `POWER: (QRP, LOW)`), all is taken verbatim.
> 
> The advantage is than any new field can be supported. But I guess this is just minor one if Cabrillo v3 stays stable.

As you moved the header to be defined by the data field it will be easier to adapt to changes in the Cabrillo version in future. So the actual proposal may be best.

> Also multiple ADDR/SOAPBOX field can be easily specified.
> 
> Option2: drop template support all together.
> 
> Option3: drop configs and keep template only.
> 
> From usability perspective I'm fine with the two level (global + contest) config without template. Of course others may have another preferences.

For me a template file would be advantageous if we could use a prefilled cabrillo log (say from last years contest) or if you had to export the log to cabrillo more than once. Than we could use the values already inserted before. But maybe that is another use case.

dl1jbe, 2020-12-30:
Thanks for the example. 

I see you used the template file for your personal information. Would it also work if you added the TEMPLATE line to your logcfg.dat instead. That way we would not have to change the rule files.

zcsahok, 2020-12-30:
Configs incl. TEMPLATE can be added to both logcfg and rule file. The difference is just the order of evaluation.

dl1jbe, 2020-12-30:
Sorry, my comment was for Ervin. He used his TEMPLATE with personal informations in the rules file.

As the personal settings should best go into logcfg.dat the TEMPLATE should be loaded there. Than we can keep the rules file unchanged for different users.

airween, 2020-12-30:
NB: I should keep the personal settings in separated template file(s). It gives more flexibility for users.

zcsahok, 2020-12-30:
We could extend the rules with the relevant Cabrillo info. (if possible)
e.g. `rules/okom`
```
...
CONTEST=okom
LOGFILE=okom.log
CONTEST_MODE
CABRILLO-CONTEST=OK-OM-DX    <<<<<<<<<<
#
MULT_LIST=okommults
COUNTRYLIST=okom:OK,OM
...
```

zcsahok, 2020-12-31:
Let me know if there are further comments/issues. If none then I'll submit a PR.

dl1jbe, 2021-01-01:
The only open point seems to be the user documentation. Otherwise go ahead with the PR.

zcsahok, 2021-01-01:
OK.

A question: would it make sense to add `/` as word delimiter for QSO format fields EXC1..4?
In the AGCW HNY contest the received rprt is '599 067/2583'.
The example Cabrillo file has this QSO line:
```
QSO:  3510 CW 2020-01-01 0909 DL9QQQ     599 001 0815    S9POL/P    599 001 4711
```
Here membership number is separated by space instead of '/'. This is not the only contest using dash: Handtastenparty has 599002/A/Heinrich/72.
Current code can only split on whitespace. The only way the achieve the expected format (without manually editing either the log or the cbr) was to extend splitting on '/' and use format CWO.
(of course I could have logged using space delimiter...)

dl1jbe, 2021-01-02:
Hmm. Either always using space AND / as delimiter or making the set of delimiters programmable by a keyword and defaulting just to space if keyword not set. I tend to make it programmable. What do you think?

zcsahok, 2021-01-02:
Yes. This way reading a Cabrillo file would reconstruct the log correctly.
Config would logically belong to the QSO format definition in `cabrillo.fmt`, something like
```
[AGCW]
EXCHANGE-SEPARATOR=/
QSO=FREQ,5;MODE,2;DATE,10;TIME,4;MYCALL,13;RST_S,3;EXC_S,6;HISCALL,13;RST_R,3;EXC1,4;EXC2,4
```
(CWO format didn't solve my problem as it has no RST)

dl1jbe, 2021-01-02:
> Config would logically belong to the QSO format definition in `cabrillo.fmt`, something like
> ```
> [AGCW]
> EXCHANGE-SEPARATOR=/
> QSO=FREQ,5;MODE,2;DATE,10;TIME,4;MYCALL,13;RST_S,3;EXC_S,6;HISCALL,13;RST_R,3;EXC1,4;EXC2,4
> ```
Right. Let us do it so.

dl1jbe, 2021-01-03:
Just an addon:

```
EXCHANGE-SEPARATOR=" \"   # Allow set of separators
```

With that we could split a comment like 

```
Hans 1273/25
```

into meaningful parts

zcsahok, 2021-01-03:
It will work exactly like that. The only (unsolvable) issue is that in this case importing a Cabrillo will use the first character from the separator string, i.e. it isn't possible to exactly reconstruct the log. But it think we can live with this.

The syntax of key-value file is somewhat tricky. Quotation marks are taken as-is. Leading space is stripped, so either one puts a trailing one which is unreadable or uses escaping like
```
EXCHANGE-SEPARATOR=\s/
```
or
```
EXCHANGE-SEPARATOR=/\s
```
Former will use space, the latter / as primary separator when reading a Cabrillo file.

dl1jbe, 2021-01-03:
Closed by commit 8367e3c. Please reopen if needed.

closed 2021-01-03 by dl1jbe