Tlf/tlf #203: Node sync/deleting QSO

m5evt, 2020-12-26:
I am running 2 instances of tlf on the same machine over LAN. I can confirm that communication between the 2 nodes is occurring and I can see a QSO being added to the other node when it takes place. However, I notice that deleting a QSO in 1 node does not delete it in the other. This results in scores drifting apart. The only way I can fix this is manually deleting the QSO in all nodes.

I shutdown one instance of tlf, added some QSOs in the other node. Then I re-opened the TLF instance again. A SYNC command (":sync") did not pick up these new QSOs that occurred while tlf was "off". Am I missing something regarding behaviour of the LAN/node implementation?

dl1jbe, 2020-12-26:
Thanks for the report.

ATM only adding QSOs to the log is supported. Normally in contest you would rather not delete a qso but make a note to process it later. So until now there was no need to support a delete also. 
Maybe we should add that too. Can you give some information what you wanted to accomplish?

An automatic sync at startup was considered to difficult to implement. So Rein, the original author of TLF, decided to provide then :SYNC command only. There were not so much need for another solution in last years as not many people seems to use TLF for M/S or M/M operation.
Besides that :SYNC should work. Only problem is that you have to provide the logfile via FTP://.

zcsahok, 2020-12-26:
@m5evt what is your `SYNCFILE` setting? (logcfg.dat)

m5evt, 2020-12-26:
Thanks for the reply. There are 2 things I am trying to accomplish that are linked. I have done some digging and I think I was missing the SYNCFILE. I will give this a try.

I have been experimenting with mapping in Python and tlf log files. I have managed to map worked cqww zones on a map using cartopy in Python.  I've also managed this for US states too. For now, this means I have to read/process the tlf log file each time I want to update the map. I had an idea that I could make the Python script emulate as a tlf node and then update the map only when a new zone is worked. Hence, I have been trying to understand the LAN functions in tlf. Below are some of the maps I have produced.

[image]

[image]

I also would like to produce real-time stats in a window (graph qso/hr, mults/hr etc.) I have some basic Python code that can do this from a tlf logfile. But again, processing a full log each time is inefficient. A database query seems the best way to do all this, so I have been experimenting with MariaDB and Python. Again, using the Python code to emulate a tlf node, then for each new QSO add it to the database. But if a QSO gets deleted, I won't pick this up. Hence, my question :)

I was trying to do this all without needing any modification to tlf.

dl1jbe, 2020-12-26:
Thanks for the reply.

As told before, besides setting the SYNCFILE in logcfg.dat you need to provide the logfile via FTP:// and have 'wget' and the common tools like 'cat', 'cp', 'sort' and 'uniq' installed.
I think in the long run we should 
a) move over to use rsync instead and
b) does a check at startup for the needed external programs.

And thanks for the explanation of your experiments. That looks quite interesting. Nice idea.
I like the idea to use the LAN protocol to feed the qsos into a database. Lets see what it takes to add a delete feature to the code. But don't hold your breath, it may take some time.

m5evt, 2020-12-26:
Thanks for the explanation. My Linux box doesn't seem to be running an ftp server. Typing "telnet localhost 21" results in "unable to connect to remote host". So trying the wget ftp:://... command manually from the command line didn't work for me. I'll have to setup some virtual machines to properly experiment with this. I agree rsync is a better way to do this.

Regarding deleting a qso; If I am in S&P mode in a very busy contest (e.g. cqww) and using ESM, sometimes by the end of the QSO I realise that I haven't successfully completed the QSO but the station has moved on to working other people. If this was a mult and I am filtering by mults on the band map (which I often do), I would incorrectly be filtering out this mult (after I had deleted the QSO). I think it is also worth adding that this LAN code is also great for an SO2R setup. I have done some basic experiments with this for SO2R and it was successful.

Happy to close the issue if you are tracking enhancements elsewhere? (I couldn't see how to label the issue as not being a bug?)

Finally, thank you to you and other devs for the hard work on tlf. It really enjoy using it and it is good to see the project active.

zcsahok, 2020-12-26:
Re rsync: it would just help transferring the log file as then it's merged using `cat log1 log2 | sort -g -k4,4 | uniq`. (Of course the merge can be written in C, but I doubt if it's worth it.)

It would be better to drop the hard coded FTP protocol and support any URL. This would allow e.g. using HTTP for which there are readily available standalone servers requiring no root access. (e.g. `python3 -m http.server 1234`)

@m5evt
Re deleting a QSO: also note that op can actually edit the log and remove/update any QSO. It's very hard to propagate such changes consistently to other nodes.

For your use case you could
- process net info to have online data (what you do now)
- and periodically (once a minute?) transfer the log (using rsync to minimize bandwidth), do local diff and update the maps accordingly.