Tlf/tlf #206: networkinginfo() needs care

dl1jbe, 2020-12-27:
Found an issue in networkinfo(): the reported config file is always either logcfg.dat or the given on command line. The system config is never shown as config_file is never empty. In addition the system config location is hard coded.
This can be fixed, but then we'll have an issue with re-reading the config via :SET/:CFG. Btw these probably anyway do not work as proper re-reading requires releasing all resources (net sockets, ...) which is not done at the moment.

dl1jbe, 2020-12-30:
@zcsahok Would it be better if we copy the default config file into the actual directory if no config are given on command line AND no logcfg.dat are already found. Furthermore close TLF with an error message if the given file is not found.

The default file normally is write protected and you have to make a copy anyway if you want to make changes.

zcsahok, 2020-12-30:
So

- if logcfg given on command line, use it or exit if not available
- if no logcfg was given
    - use locfg.dat from current dir if available
    - else copy from share and open it
    - else (if copy/open fails) exit
 
Makes sense.

Referenced data files (rules, exchanges) are first tried from current dir (=the one where Tlf was started from) then from share. There shall be a common function for this.

dl1jbe, 2020-12-31:
See https://github.com/dl1jbe/tlf/tree/copy_defconfig for a possible implementation.

dl1jbe, 2021-01-12:
Problem solved in PR #215. Closing the issue.

closed 2021-01-12 by dl1jbe