Tlf/tlf #211: src/sendbuf.c:47: multiple definition of `buffer'; test_adif.o:./test/test_adif.c:63: first defined here

df7cb, 2021-01-05:
Before submitting #210 I was trying to see if it was already fixed in git, past 1.4.1, but it seems the "-fno-common" problem is back in git HEAD, originally fixed in 7b3c13f868934e:

```
gcc -Wall -g -O2 -fdebug-prefix-map=/home/myon/tlf.git=. -fstack-protector-strong -Wformat -Werror=format-security  -Wl,-z,relro -Wl,-z,now -o run_bands run_bands.o test_bands.o data.o functions.o ../src/bands.o -lm -lglib-2.0 -lhamlib -lpanelw -lncursesw -ltinfo -lcmocka -Wl,-wrap=sleep -Wl,-wrap=key_get -Wl,-wrap=key_poll -Wl,-wrap=sendto -Wl,-wrap=wgetch -Wl,-wrap=refreshp
/usr/bin/ld: ../src/sendbuf.o:./src/sendbuf.c:47: multiple definition of `buffer'; test_adif.o:./test/test_adif.c:63: first defined here
```

dl1jbe, 2021-01-06:
Thanks for the report Christoph. Will be fixed shortly

dl1jbe, 2021-01-06:
Should be fixed with commit e83a7528c9787507569f458db30cb2ed0ae24933.
Please reopen if problem still exists.

closed 2021-01-06 by dl1jbe