Tlf/tlf #219: Fldigi in Debian Bullseye is dependent on libflxmlrpc

N0NB, 2021-01-17:
Getting back into working with Tlf a bit I decided to enable the XMLRPC configure option and it failed.  So I installed Fldigi and saw that it is dependent on `libflxmlrpc` and also installed the `libflxmlrpc-dev` package manually.  Of course, `configure` failed as at the very least the name of the library has changed.  After staring at `configure.ac` for a few minutes I looked at the symbols in the new library and found that none of the symbols `AC_CHECK_LIB` will look for are present.  Then I checked the `doc` directory and found `/usr/share/doc/libflxmlrpc-dev/README` which states:

```
flxmlrpc Library

This is version 0.1 of flxmlrpc, an implementation of the XmlRpc protocol
written in C++, based upon XmlRpc++0.7 and modified to provide additional XmlRpc
Variable types.  It is used in fldigi, flrig, flnet, flmsg, flarq, flamp, fllog;
a suite of programs written for amateur radio emergency communications.

flxmlrpc is designed to make it easy to incorporate xmlrpc client and server
support into C++ applications. Or use both client and server objects in your
application for easy peer-to-peer support.

Features

    Easy     This library is easy to incorporate into C++ applications. No other
             libraries are required, other than your system's socket libraries.
             Simple XML parsing and HTTP support are built in.

    Fast     All IO is non-blocking, so a slow client or network will not slow
             down the server.

    Portable Written in standard C++ to the POSIX and Windows sockets APIs.
             You do need a fairly recent compiler, g++ 3.1 or greater.

    Free     This library is released under the GNU LGPL.
```

So, this is a C++ library and it says it is easily incorporated into a C++ program, but Tlf is written in C.  Is this a problem?

As a note, the `tlf` package in Debian Bullseye is dependent on `libxmlrpc-core-c3`.  I have not checked if Tlf and Fldigi can communicate with the two different libraries as I've never set up the XMLRPC interaction between the two.

As this is well above my pay grade (ha!), I'm filing this issue to give everyone a heads up on this.

airween, 2021-01-17:
Hi Nate,

sorry to tell you but I can't reproduce your issue.

I have an up-to-date Bullseye:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux bullseye/sid
Release:	testing
Codename:	bullseye
```

Grabbed Tlf and compiled first - then I didn't have `libflxmlrpc-dev` package, only the required libraries (Hamlib and libxmlrpc-core-c3, libncurses, glib, ...):
```
$ autoreconf --install
configure.ac:16: installing './compile'
configure.ac:45: installing './config.guess'
configure.ac:45: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
src/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
$ ./configure --enable-fldigi-xmlrpc
...
----------------------------------------------------------------------

 Tlf Version 1.5~git configuration:

 Prefix         /usr/local
 Preprocessor   gcc -E 
 C Compiler     gcc -g -O2

 Package features:

 With XML RPC	true
```

Tlf had successfully compiled.

Then I started to install fldigi from source - installed all required libraries, and I got:
```
$ ./configure
...
Configuration summary:

  Version ..................... 4.1.17

  Static linking .............. no
  CPU optimizations ........... none
  Debugging ................... no

  fldigi ...................... yes
  flarq ....................... yes

  i18n ........................ yes

 fldigi build options:

  oss ......................... yes
  portaudio ................... yes
  pulseaudio .................. no

  libmbedtls .................. internal lib
  flxmlrpc .................... internal lib

  hamlib ...................... yes
```

Note: the `flxmlrpc` is showed as "internal".

Then I installed `libflxmlrpc-dev` package, but Tlf was still able to built (after `make clean`, `./configure --enable-fldigi-xmlrpc` and `make`).

I ran again the Fldigi's `configure` script, and I got:
```
Configuration summary:

  Version ..................... 4.1.17

  Static linking .............. no
  CPU optimizations ........... none
  Debugging ................... no

  fldigi ...................... yes
  flarq ....................... yes

  i18n ........................ yes

 fldigi build options:

  oss ......................... yes
  portaudio ................... yes
  pulseaudio .................. no

  libmbedtls .................. internal lib
  flxmlrpc .................... external lib

  hamlib ...................... yes
```
(as it seems the `flxmlrpc` became as external)

What do I forget? Could you help me (eg. step-by-step), how can I reproduce this issue?

N0NB, 2021-01-17:
My point is, now that Fldigi is using its own internal XMLRPC, and Tlf is using the more commonly available one, wll the XMLRPC communication work between the two applications?  As I've never set that up, I am raising this issue for others to test.

As Tlf and Fldigi now use separate XMLRPC libraries, can they still communicate?  I hope someone who uses this functionality can test.

dl1jbe, 2021-01-30:
@airween As you are the one working with the digi-modes most here, do you have a chance to have a look into Nates question?

(Unrelated to this: Are you common with Debians release planning? When should a new TLF release be ready to be picked up for next Debian release?)

N0NB, 2021-01-30:
As I understand it, Debian entered a "soft freeze" a couple of weeks back.  I'm not sure if a new release of Tlf would make it in if it's considered anything other than a bug fix release.  I don't track Debian policy all that closely.

airween, 2021-01-30:
> @airween As you are the one working with the digi-modes most here, do you have a chance to have a look into Nates question?

ah, sorry - were very busy days here... Sure, I'll look at the flxmlrpc usability for Tlf soon.

> (Unrelated to this: Are you common with Debians release planning? When should a new TLF release be ready to be picked up for next Debian release?)

As I know the current Debian testing is coming soon as stable, here is the [timing](https://release.debian.org/bullseye/freeze_policy.html).

May be @df7cb can helps in this question: Christoph, can we put the new version before 02.12 to Debian Testing? I'm afraid we can't. Note, the next stable Debian will contain the [1.4.1](https://packages.debian.org/bullseye/tlf).

dl1jbe, 2021-01-30:
Thanks to both of you.

> As I know the current Debian testing is coming soon as stable, here is the [timing](https://release.debian.org/bullseye/freeze_policy.html).
>
> May be @df7cb can helps in this question: Christoph, can [...]

Ah, that was the GCC-10 thing we sorted out some days ago. Ok, than we are a little bit late. I fear the 1.4.1 will have to go in. As you are more involved with Debian Ervin maybe you can give us a hint for coming releases next time. 

> > @airween As you are the one working with the digi-modes most here, do you have a chance to have a look into Nates question?
> 
> ah, sorry - were very busy days here... Sure, I'll look at the flxmlrpc usability for Tlf soon.

No problems here. I just wanted to ping if someone else should look into it. Would be fine if you can find time in next weeks.

airween, 2021-01-30:
After a quick look, I'm afraid `flxmlrpx` isn't compatible with `xmlrpc`:

```
$ gcc -Wall flxmlrpctest.c -o flxmlrpc -lflxmlrpc
/tmp/ccD564fF.o: In function `main':
flxmlrpctest.c:(.text+0xa): undefined reference to `xmlrpc_read_int'
collect2: error: ld returned 1 exit status
```

I'm going to search `flxmlrpc` library API.

airween, 2021-01-30:
This is what I [found](http://www.w1hkj.com/files/flxmlrpc/readme.txt):

_This is version 0.1.4 of flxmlrpc, an implementation of the XmlRpc protocol written in **C++**, based upon XmlRpc++0.7 and modified to provide additional XmlRpc Variable types._

So please correct me if I'm wrong, but I think if we want to use this library (in C), we have to implement a wrapper - but that will need also a C++ compiler. Or we can use it as native, with C++ compiler directly. Do we want this?

N0NB, 2021-01-30:
If the actual XML RPC communications mechanism would be compatible between the libraries, then there really is not issue, but if the version Tlf relies on cannot accept the message from the Fldigi version, then that needs to be addressed, IMO.

dl1jbe, 2021-01-30:
As long as the protocol is compatible I see no need to use `flxmlrpc`. If the extensions there are needed hast to be checked.

As fldigi is known to communicate with a quite some different applications I assume that there is quite some commonality in the protocols.

Wrt the use of c++ libs: Technically it is possible to use a c++ library from c code, but makes some more work.

airween, 2021-01-30:
I don't have Debian Bullseye with desktop, so I can't check the versions - but tested on a Mint.

```Configuration summary:

  Version ..................... 4.1.18

  Static linking .............. no
  CPU optimizations ........... none
  Debugging ................... no

  fldigi ...................... yes
  flarq ....................... yes

  i18n ........................ yes

 fldigi build options:

  oss ......................... yes
  portaudio ................... yes
  pulseaudio .................. no

  libmbedtls .................. internal lib
  flxmlrpc .................... external lib

  hamlib ...................... yes
```

and Tlf:

```
 Tlf Version 1.5~git configuration:

 Prefix         /usr
 Preprocessor   gcc -E 
 C Compiler     gcc -g -O2

 Package features:

 With XML RPC	true
```

I think the `libxmlrpc` is compatible with `flxmlrpc`. I grabbed the `fldigi-shell` Perl script from [here](http://www.w1hkj.com/files/fldigi/fldigi-shell), which uses Perl XML::RPC lib (which is fully compatible with xmlrpc-c).

```
$ ./fldigi-shell 
fldigi % modem.get_name
RTTY
fldigi % rx.get_data
[I got what I saw in RX window]
```

But something is wrong in Tlf code: it can't communicate with Fldigi, but I don't think that the problem is something interoperability issue. I ran `tcpdump` and the communication is well between Tlf and Fldigi. Nonetheless after a short time I got `Fldigi: lost connection!` message at bottom left corner. The last request and response were:

```
POST /RPC2 HTTP/1.1
Host: 127.0.0.1:7362
Accept: */*
Content-Type: text/xml
User-Agent: Xmlrpc-c/1.33.14 Curl/7.58.0
Content-Length: 191

<?xml version="1.0" encoding="UTF-8"?>
<methodCall>
<methodName>rig.set_frequency</methodName>
<params>
<param><value><double>14089825</double></value></param>
</params>
</methodCall>
```

```
HTTP/1.1 200 OK
Server: XMLRPC++ 0.8
Content-Type: text/xml
Content-length: 142

<?xml version="1.0"?>
<methodResponse><params><param>
	<value><double>14089826,000000</double></value>
</param></params></methodResponse>
```

I'll check that soon.

N0NB, 2021-01-30:
Thanks for that test, Ervin.

airween, 2021-01-30:
Looks like the `flxmlrpc` isn't fully compatible with `xmlrpc-c` - see the communication my last comment.

The server response contains wrong (or wrong formatted) value: `<double>14089826,000000</double>` - IMHO this isn't a double format.

Same effect with fldigi-shell:
```
$ ~/bin/fldigi-shell -d -c "rig.set_frequency 14090824.000000"
Method rig.set_frequency takes 1 args
$VAR1 = [
          'rig.set_frequency',
          '14090824.000000'
        ];
Encoding arg 1 as d
Error: RPC::XML::Client::send_request: parse-level error: Bad floating-point data read at document line 3, column 31 (byte 87)
```
Fldigi allows the request (`<double>14089825</double>`) and sets the frequency value on its own field, but the library can't parse the response. Tlf issue triggered [here](https://github.com/Tlf/tlf/blob/master/src/fldigixmlrpc.c#L276), when it calls `rig.set_frequency`.

I also checked Fldigi's xmlrpc inside code, looks like it uses a simple `%f`, so I'm confused why we get `14089826,000000` format....

I'm going to make a report to Dave.

zcsahok, 2021-01-30:
What locale are you using?

see https://stackoverflow.com/questions/48968409/printfs-floating-point-format-flag-f-for-english-number-format-only

airween, 2021-01-30:
Bingo.

A simple `export LANG=C` before start Fldigi solved the problem - Tlf works as well.

I think we successfully walked around the issue :).

Thanks for everyone for help and for question.

df7cb, 2021-02-01:
> As I know the current Debian testing is coming soon as stable, here is the [timing](https://release.debian.org/bullseye/freeze_policy.html).
>
> May be @df7cb can helps in this question: Christoph, can [...]

The actual deadline is Feb 12th, but that's after the migration delay from unstable to testing, which is usually 5 days until that deadline. So there is still time until Feb 6th or maybe 7th, but I wouldn't recommend rushing things unless they are already ready.

N0NB, 2021-02-01:
* On 2021 01 Feb 11:49 -0600, Christoph Berg wrote:
> The actual deadline is Feb 12th, but that's after the migration delay
> from unstable to testing, which is usually 5 days until that deadline.
> So there is still time until Feb 6th or maybe 7th, but I [...]

How about the possibility of using Debian backports?  I'm running
Bullseye (Testing) so I've no need for that repository at the moment.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

df7cb, 2021-02-01:
Sure, we can always upload the new version to bullseye-backports later.

dl1jbe, 2021-02-02:
Well ok thanks. As there is quite some work to do for a stable release I think we should not make haste.
If we can provide a backport later - say in 4..6 months - that would be good.

dl1jbe, 2021-03-01:
@Nate: Do you see some open questions here Nate? If not, please close the issue.

N0NB, 2021-03-01:
I think so long as interoperability exists between Tlf and FLdigi, then this is all I wanted the group to be assured of.

closed 2021-03-01 by N0NB