Tlf/tlf #235: Mults on bandmap with custom rules

m5evt, 2021-01-30:
I am using these rules:

https://lists.nongnu.org/archive/html/tlf-devel/2016-01/txtVP8nDiH35g.txt

for cqww 160. The bandmap doesn't mark mults and so I can't filter "only mults" on the bandmap. Is there anything I can change in the rules file to get this to work? I am using the release tagged as tlf-1.4.1

dl1jbe, 2021-01-30:
Sorry, at the moment "only mults" works only for hardcoded CQWW contest. You could change the CONTEST=cqww160 definition in your rules file, but that will loose the custom point scoring.

m5evt, 2021-02-02:
Thanks, I had a quick look at the source code to try and hack the function `bm_ismulti()` in bandmap.c but it didn't work, so I took your suggestion of changing the rules to cqww.

I have a vague memory that this worked as expected for wpx.

For a low power S&P assisted contester this feature working for all mults would be great. I spend a lot of time moving between ctrl+g and alt+g.

dl1jbe, 2021-02-02:
`bm_ismulti() ` is a long standing project which needs to be extended in next time.
We did a first step with the cqww multis some time ago, but did not fount time to  make it useful for other contests. Sorry.

m5evt, 2021-02-05:
Thanks for the reply. No need to say sorry, I completely understand the problem with finding time.

PR #386 (2023-05-11): Add CQ WW 160
Aiming to resolve #235.

First aligned the naming of `spot.band` with it actual use: it stores a band index (0,1,...), not a band (160,80,...). This was marked earlier in `qso_from_spot()`, where a temporary QSO is created for checking it for a new mult. Now `qso->band` contains the human readable band value (160,...).
Naming of `freq2band()`  in `bands.c` was also fixed. (Sometime band index is referred to as band number, not easy to follow)

Next added setting of `band` value for a newly entered QSO. It is correctly filled in when reading the log, but was left out for new ones. [... cut]
closed 2023-05-11 by zcsahok