Tlf/tlf #252: DX_&_SECTION keyword not working as documented

dl1jbe, 2021-03-05:
Man page states that with DX_&_SECTION keyword ` Multiplier is DXCC country or section from multiplier file`

In last weeks there were at least two reports from DL3EAZ and F8CFE which states that the keyword does not work in that manner. A closer look to the code showed that it is tightly coupled to ARRL contests which often use a multiplier scheme where W or VE station can get multipliers for W/VE sections or other DXCC countries. So the keyword in its current form is useless for other contests like EUDX or `Coupe du REF`.

The current implementation of DX_&_SECTIONS works as follows:

If Country is W or VE do not count as country multiplier, all other countries are mults
If Country is W or VE do count sections as multiplier, for all other countries ignore it.

So country and section multiplier are mutual exclusive . That is also the case for `Coupe du REF`.
Some other contests (e.g. EUDX) count country AND multiplier for the EU countries.

I think a way to solve the problem can be the following:

- Rename DX_&_SECTION to ARRL_DX_&_SECTIONS keeping the behaviour and adding a new DX_&_SECTION which simply counts DXCC countries and sections both as multipliers. If you have to exclude your own country from getting country points (see `Coupe du REF`) use COUNTRYLIST=<your countries> and EXCLUDE_COUNTRYLIST.

- The new DX_&_SECTIONS together with EXCLUDE_COUNTRYLIST can also handle ARRLDX contests if it is guaranteed that no stations outside of W and VE uses sections in their reports.

Before starting to implement  solution some feedback would be good.

zcsahok, 2021-03-06:
Good question. In addition to multiplier handling there was also a complaint that mult screen is empty. So changing the logic would just fix the actual mult counting. Renaming the config would interfere with existing logcfg/rules (assuming this option is used).

I would simply update the documentation and focus on getting it covered by the plugin approach (i.e. adding generic mult handling). But this is just my 2 cents.

zcsahok, 2023-08-14:
As **DX_&_SECTION** came up for Portugal Day contest in #346 I made a quick check of contest rules.

| Contest | Multipliers |  Section/DXCC clash |
| --- | --- | --- |
|--_legacy_-- ||
|ARRL DX |        W/VE sections or DXCCs  |    yes, solved by internally hardcoded logic |
|--_exclusive_--||
|Coupe du REF  |  F depts + non-F DXCCs   |    no | 
|HADX     |       HA counties + non-HA DXCCs  | yes |
|--_inclusive_--||
|EUDX     |       EU regions + DXCCs   |       n/a |
|Portugal Day  |  5*(CT districts) + DXCCs   | n/a |

Contests with _exclusive_ multipliers can be modeled with the single multiplier support of TLF. If there is a clash between sections and DXCCs, then sections can be prefixed to make them unqiue. This is done for example in case of HADX: all counties get a HA/ prefix to avoid the clash between HA county HB and Switzerland.

_Inclusive_ contests -- provided that they follow the N*(own mults)+DXCCs logic -- can be relatively easiely supported by adding worked DXCCs to the total number of multipliers in `get_nr_of_mults()`.
Log line could accomodate the extra multiplier info like this (for EUDX):
```
5    6         7         8
56789012345678901234567890
exchange      mult    pp
AT01     OE.. AT01    10
AT02     .... AT02    10
1234     PY.. ....     5
```
(dots just show field width)

One way of configuring it could be 
```
MULT_DXCC
MULT_WEIGHT=5
```

Even if we have rules for it, the **ARRL 10m** contest is not fully supported. It has simular rules to ARRL DX, but multpliers are counter per mode (CW/SSB). In addition to this, "local" countries also include Mexico. So DX_&_SECTION won't work here.