Tlf/tlf #267: FILPATHLEN become HAMLIB_FILPATHLEN in hamlib 4.2

jackhill, 2021-06-05:
FILPATHLEN was changed to HAMLIB_FILPATHLEN starting in [hamlib 4.2](https://github.com/Hamlib/Hamlib/blob/ff9a745c7a716791d58932ed6b82036823094ab7/NEWS#L40), so tlf does not currently build against hamlib 4.2. The error message as expected is `sendqrg.c:103:6: error: ‘FILPATHLEN’ undeclared (first use in this function)`. Presumably the fix is to update `sendqrg.c` to use the new macro. I don't see any pre-processor conditions elsewhere in the code, but one might need to be introduced here to maintain compatibility with older hamlib versions as well.

airween, 2021-06-05:
Hi @jackhill,

thanks for the hit - could you try [this](https://github.com/airween/tlf/tree/issue267) [quick](https://github.com/airween/tlf/commit/851321c0979ce81d4d69d1a907cf3c6ecf23d1c9) fix?

@dl1jbe, @zcsahok - do we need to handle the other possibilities in [sendqrg.h](https://github.com/airween/tlf/commit/851321c0979ce81d4d69d1a907cf3c6ecf23d1c9#diff-df6bdd285e00547aeb88be83e820f45c5d345c9ecc97d745f594276812b9a04bR26-R32)?

jackhill, 2021-06-05:
@airween Thanks for looking into it! I can confirm that I can build and run the tests with your branch and hamlib 4.2. I didn't test running it and connecting it to any radios (I'm working on creating a package of tlf for [Guix](https://guix.gnu.org) and there are some unrelated things for me to clean up before I can do that test) or against an older hamlib.

de KM4MBG

airween, 2021-06-05:
Hi @jackhill,

thanks for confirmation. Please do not apply this patch in Guix while we do not merge into the master branch.

73, de HA2OS

jackhill, 2021-06-06:
Sounds reasonable. I'll wait to see what the final solution is and backup that fix or package a newer release when one if available. Thanks again!

dl1jbe, 2021-06-06:
> Hi @jackhill,
>
> thanks for the hit - could you try [this](https://github.com/airween/tlf/tree/issue267) [quick](https://github.com/airween/tlf/commit/851321c0979ce81d4d69d1a907cf3c6ecf23d1c9) fix? [...]

We should at least do a '#else' and '#error' after line 30 if neither macro is defined.

The whole situation with Hamlib breaking its public API (using FILPATHLEN was a recommended part of it) in a minor version bump is highly disappointing There are quite some more ham radio applications which will be affected by that move.

While we can fix it for new code it is still a problem for using hamlib-4.2 with existing released versions of TLF. Hopefully we can work around by using '-DFILPATHLEN=HAMLIB_FILPATHLEN' or similar during build.

airween, 2021-06-06:
Hi Thomas,

I thought about I put the check into the autoconfig script, but using of `FILPATHLEN` macro is very limited: it's only in the `sendqrg.c` (and the header now). That's why I chose this solution. Would you me to put it the `configure.in`?

dl1jbe, 2021-06-06:
Hi Ervin, think it should not be needed.  We check for hamlib in configure.in and so either macro should be here. If not, then something is quite wrong and we should '#error' out in sendqrg.h

airween, 2021-06-06:
Thanks - did you think something like [this](https://github.com/Tlf/tlf/compare/master...airween:issue267?expand=1)?

If it's right, I'm going to send the PR.

dl1jbe, 2021-06-06:
Better

#error "(HAMLIB_)FILPATHLEN  macro not found"

to catch both versions.
Otherwise feel free to PR and merge it in.

airween, 2021-06-07:
PR #268 has merged, it solves this issue.

PR #268 (2021-06-07): Fix FILPATHLEN macro handling
Fixes #267.
closed 2021-06-07 by airween