Tlf/tlf #273: Mult is added in scoring window when no new mult is worked

N0NB, 2021-08-29:
Yesterday I noticed that seemingly at random the score window would increment the mult count when no mult was worked and the log window correctly showed that no mult was worked.  Using `:REScore` or restarting Tlf resulted in the log file being read and the mult count and score being corrected,

From my rules file:

```
#################################
#      Scoring rules            #
#################################
#
MIXED
RECALL_MULTS
SSBPOINTS=2
CWPOINTS=3
SECTION_MULT_ONCE
### WYSIWYG_ONCE
MULT_LIST=ksqp.txt
```

I will note that I had erroneously had `SECTION_MULT` instead and after stopping Tlf, adding `_ONCE`, manually editing out the erroneous section mults, and restarting Tlf, the mult count in the scoring window was correct.

I did not observe that incrementing the mult count in the scoring window was related to `RECALL_MULTS` as it seemed to do it for both recalled callsigns and stations not worked before.

I've attached the section mults file.

[ksqp.txt](https://github.com/Tlf/tlf/files/7072206/ksqp.txt)

zcsahok, 2021-08-29:
I'm looking at multiplier handling at the moment, so may find something related to this issue.
What was your `CONTEST` setting?
By "log window" do you mean the log line showing/not showing the worked mult?

N0NB, 2021-08-29:
Hi Zoli.

Here is the contest section:

```
########################
# General contest mode #
########################
#
CONTEST=ksqp
LOGFILE=ksqp_2021.log
CONTEST_MODE
CABRILLO=UNIVERSAL
CALLMASTER=MASTER.SCP
```

I noticed that the issue doesn't seem to occur as often in SSB mode.

N0NB, 2021-08-30:
At the end of the party I ran a full hour of CW and the mult value in the scoring window was rather stable.

By "log window" I mean the mini-log that appears above the entry line.

zcsahok, 2022-08-28:
Did I get it right, that you have experienced that mults kept "randomly" added after logging a CW QSO?
Was the log line correctly showing or not showing the mult? I.e. only the total sum was wrong? Or both were off?

Could you please post your configs?

N0NB, 2022-08-29:
The log line was correct but the scoring window was way off.

Rules file `ksqp`:

```
########################
# General contest mode #
########################
#
CONTEST=ksqp
LOGFILE=ksqp_2022.log
CONTEST_MODE
CABRILLO=UNIVERSAL
CALLMASTER=MASTER.SCP
#
##################################
#                                #
#  Messages F1= to F12=          #
#  Message CQ_TU_MSG=            #
#  Message S&P_TU_MSG=           #
#                                #
#  % = call                      #
#  @ = hiscall                   #
#  # = serial                    #
#  [ = RST                       #
#  + = increase cw speed         #
#  - = decrease cw speed         #
#                                #
##################################
#
F1=CQ KQP %
F2=@ DE %
F3=@ 5NN MSH
F4= TU
F5=@
F6=%
F7=@ SRI QSO B4 GL
F8=AGN
F9=?
F10=QRZ?
F11=PSE K
F12=CQ KQP %
#
CQ_TU_MSG=TU %
S&P_TU_MSG=TU 5NN MSH
#
#ALT_0=
ALT_1=DE N0N, DE N0NB OP
#ALT_2=
#ALT_3=
#ALT_4=
#ALT_5=
#ALT_6=
#ALT_7=
#ALT_8=
#ALT_9=
#
#SEND_DE
#
#################################
#                               #
#  Voice Keyer Files            #
#    (F1 to F12)                #
#################################
#
VKM1=/home/nate/logs/wav/N0N/kqp/F1.wav
VKM2=/home/nate/logs/wav/N0N/kqp/F2.wav
VKM3=/home/nate/logs/wav/N0N/kqp/F3.wav
VKM4=/home/nate/logs/wav/N0N/kqp/F4.wav
#VKM5=
VKM6=/home/nate/logs/wav/N0N/kqp/F6.wav
#VKM7=
VKM8=/home/nate/logs/wav/N0N/kqp/F8.wav
#VKM9=
VKM10=/home/nate/logs/wav/N0N/kqp/F10.wav
VKM11=/home/nate/logs/wav/N0N/kqp/F11.wav
VKM12=/home/nate/logs/wav/N0N/kqp/F12.wav
VKSPM=/home/nate/logs/wav/N0N/kqp/SPM.wav
VKCQM=/home/nate/logs/wav/N0N/kqp/CQM.wav
#
#
#################################
#      Scoring rules            #
#################################
#
MIXED
RECALL_MULTS
SSBPOINTS=2
CWPOINTS=3
SECTION_MULT_ONCE
MULT_LIST=ksqp.txt
####### END #####################
```
The `logcfg.dat` file:

```
RULES=ksqp
#
#################################
#                               #
#   TLF-LOGCFG.DAT v. 1.1.0     #
#                               #
#  Uncomment the options you    #
#  want to enable. See tlf.doc  #
#  for a description of the     #
#  options. You can keep diff-  #
#  erent versions for different #
#  contests. I keep separate    #
#  configuration files for      #
#  each contest. If you enable  #
#  more than 1 mutually exclu-  #
#  sive options, the last one   #
#  will be efective.            #
#                               #
#                               #
#################################
#
#CTCOMPATIBLE
#
#################################
#                               #
#   EDITOR                      #
#                               #
#################################
#
#EDITOR=joe
EDITOR=vim
#EDITOR=e3
#EDITOR=mcedit
#
#################################
#                               #
#  CALL                         #
#                               #
#################################
#
CALL=N0N
#
#
#################################
#                               #
#  Time offset from UTC         #
#                               #
#################################
#
TIME_OFFSET=0
TIME_MASTER
#
#################################
#                               #
#  LAN PORT                     #
#                               #
#################################
#  addnode only OTHER nodes !!
#
#ADDNODE=10.0.0.115
#ADDNODE=192.168.1.2
#
THISNODE=A
#
LAN_DEBUG
#
#################################
#                               #
#  KEYERPORT                    #
#                               #
#################################
#
NETKEYER
NETKEYERPORT=6789
NETKEYERHOST=127.0.0.1
#
#################################
#                               #
#  KEYERPARAMETERS              #
#                               #
#################################
#---speed (6 ... 60 wpm)
CWSPEED=24
#---weight (-5 ... 5 ms)
WEIGHT=1
#---cq delay (in 0,5 s)
CQDELAY=10
#---txdelay (ms)
TXDELAY=2
#---sidetone (200...800, 0 = mute)
CWTONE=800
#################################
#                               #
#  PACKET INTERFACE             #
#                               #
#################################
# Cluster:
TELNETHOST=
TELNETPORT=
# automatic login ###############
CLUSTERLOGIN=
# write log to disk #############
CLUSTER_LOG
#
# use tnc instead of telnet #####
#TNCPORT=/dev/ttyS0
#TNCPORT=/dev/ttyUSB1
#TNCSPEED=2400
#
# get clusterinfo from network ##
#FIFO_INTERFACE
#
#################################
#                               #
#  RADIO CONTROL                #
# (comment out if not present)  #
# Rigmodel = Hamlib index, here #
#           for ten tec OMNI VI #
#################################
#
RADIO_CONTROL
RIGMODEL=2029
RIGSPEED=38400
RIGPORT=/dev/rig
RIGPTT
#
#SSBMODE
#
#RIT_CLEAR
#
#SHOW_FREQUENCY
#
#################################
#                               #
#  INFORMATION WINDOWS          #
#                               #
#################################
#
#CLUSTER
BANDMAP=S,900  # skipdupes, 900s livetime
#FILTER
#
SCOREWINDOW
#
CHECKWINDOW
#################################
#                               #
#  CHECK PARTIAL FUNCTION       #
#                               #
#################################
#
PARTIALS
#USEPARTIALS
#
#################################
#                               #
#  CONDX (info for muf calc.)   #
#                               #
#################################
#
SUNSPOTS=40
SFI=80
#
#################################
#                               #
#     Misc commands             #
#                               #
#################################
# No automatic 'qso before msg'
NOB4
#
# No automatic cq
#NOAUTOCQ
#
# Left/right arrows don't change band...
#NO_BANDSWITCH_ARROWKEYS
#
# Marker file for Xplanet
MARKERS=tlfmarkers
#
#############END#################
```

N0NB, 2022-08-29:
I have a Perl script I wrote some time back that processes the Cabrillo file to look for discrepancies:

```
$ ./log-calc.pl

Number of mults: 45

Mults/QSOs:
AL:   4  AR:   8  AZ:   8  CA:  30  CO:   4  CT:   2  DE:   4  DX:   5  
FL:  18  GA:  12  HI:   1  IA:   4  IL:  13  IN:   2  KS:  12  KY:   7  
MA:   7  ME:   2  MI:  11  MN:  10  MO:   8  MT:   1  NB:   1  NC:   6  
ND:   2  NH:   2  NJ:   8  NM:   1  NV:   7  NY:  16  OH:  16  ON:   6  
OR:   3  PA:  15  SC:  11  SD:   2  SK:   4  TN:  11  TX:  11  VA:  12  
VT:   2  WA:   8  WI:   6  WV:   5  WY:   1  

80m CW: 22        SSB: 0         Digital: 0         
40m CW: 106       SSB: 3         Digital: 0         
20m CW: 142       SSB: 26        Digital: 0         
15m CW: 16        SSB: 0         Digital: 0         
10m CW: 0         SSB: 0         Digital: 0         

Totals
All: 329  Duped: 315  
CW: 286  SSB: 29   Digital: 0    

Points:
Pts CW: 858         SSB: 58          Digital: 0           Total: 916

Total claimed score: 41220
```

In contrast, here is what TLF is showing:
[image]

zcsahok, 2022-08-29:
Using #342 I get the correct points. Mults are still wrong, of course.
[image]

zcsahok, 2022-08-29:
(somehow my previous comment was lost, so again)

Using your inputs I was able to roughly reconstruct your log and observe a similar behavior.
[image]

There are at least two issues.

1. re-scoring does not consider the actual QSO mode, instead it takes the currently set mode (usually CW). This explains the 945 as it's 315*3 (all CW points). I've opened an issue for this (#341) and provided a fix (#342).
2. mult count is wrong: there are only 45 different mults in the log. This still has to be investigated, but at least it's reproducible (kind of).

N0NB, 2022-08-30:
Zoli, if you would like a copy of my actual log to work with, I can email it privately.  In it are QSO records that are from in-state operators with a three letter county exchange, although I didn't really notice a pattern of extraneous mults involving them.

zcsahok, 2022-08-30:
Applying both fixes produces the expected result:
[image]

In this case it was instrumental to have the re-scoring feature in place as it allowed easily reproduce the issue. Without it one would have had to deduce the error pattern by manual log entry.

zcsahok, 2022-09-01:
Oops, I didn't to close this but merging #343 did it. Reopening until final OK.

zcsahok, 2022-09-07:
@N0NB Nate, did you have a chance to check if all is OK now?
Could we also add the rule file to the repo?

N0NB, 2022-09-07:
I have not tested this.  If it does a rescore it should be correct?

N0NB, 2022-09-07:
Okay, just did test it and it rescored correctly it appears!

[image]

As for the rules file, adding it in is fine.  It would probably be best to rename `ksqp-instate` or some such.  Others might not like my files paths and such so they can edit it!

I think this can be closed.

closed 2023-04-12 by dl1jbe