Tlf/tlf #277: String given to CABRILLO-EXCHANGE truncated after ten characters

N0NB, 2021-09-08:
As noted in this [message to the list](https://lists.nongnu.org/archive/html/tlf-devel/2021-09/msg00007.html), CABRILLO-EXCHANGE is apparently limited to ten characters.  I had set up an NA QP Cabrillo format as:

```
# used for North American QSO Party, Colorado QSO Party
[NAQP]
QSO=FREQ,5;MODE,2;DATE,10;TIME,4;MYCALL,15;EXC_S,14;HISCALL,15;EXC1,10;EXC2,3
```

to align with the [CO QP rules](http://ppraa.org/downloads/coqp/COQP%20Rules.pdf) that state that fields must start on the designated columns as shown in the [NA QP example](http://ncjweb.com/NAQP_Cabrillo_Template.txt) (Rule 11 i.):

```
QSO: 28048 CW 2014-01-11 1800 N5KO            TREY       CA  N4PN            PAUL       GA
```

(apparently programmers can no longer split fields on white space).

In an effort to pad the area between my name and state I added enough spaces in the argument to CABRILLO-EXCHANGE so the string length was 14 characters to match the `EXC_S,14` in the format file, except that at after ten characters the string was truncated.

Fortunately, with only 22 Qs and Vim, manually adjusting the file was possible.

It seems to me that increasing this to 16 or 32 characters should do no harm as the format should be the final arbiter of the field width.

Thoughts?

N0NB, 2021-09-08:
One thought I had is the addition of EXC_S1, EXC_S2, etc. specifiers to the Cabrillo format lexicon.  It would then be possible to set field lengths for each element independently.

dl1jbe, 2021-09-14:
Nate, can you please check that the issue got fixed by #278? If so, please close the issue. Thanks

N0NB, 2021-09-14:
* On 2021 14 Sep 12:27 -0500, Thomas Beierlein wrote:
> Nate, can you please check that the issue got fixed by #278? If so,
> please close the issue. Thanks

I will do so, Tom.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

dl1jbe, 2021-11-15:
No feedback, closing. Seems to be solved.

closed 2021-11-15 by dl1jbe

N0NB, 2021-11-15:
Sorry, Tom.  I must have gotten distracted and forgot about this.  I will *this time* try to check it along with a couple of recent merges.

Oft times I need some reminders.  Sigh.

dl1jbe, 2021-11-15:
Never mind, Happens here too.