Tlf/tlf #280: regressions with the actual master

dh5fs, 2021-10-17:
1. when using <SHIFT-F1> if you want to return to your CQ frequency (in MEM) after S&P some Calls are filled in, expectation would be to have a clear call input field
2. when changing CW speed and a call is in the input field, some letters are added there, expection: no letters anywhere when using <PG-UP, PG-DWN> for changing CW speed

dh5fs, 2021-10-17:
3. one more: there seems to be a new directory structures since some time. I always install with the --prefix option, after that actual TLF does not find the eg. rules directory, which needs to be now under /bin/rules, but was before under /share/tlf/rules. Same is true for logcfg.dat, cty.dat etc. they are not found. So basically install with --prefix does not work properly.
Actually the new structure is better, but does not get installed.

dl1jbe, 2021-10-19:
PgUp/PgDwn problem got fixed in master branch

dh5fs, 2021-11-14:
issues #1 and #2 seem to be fixed, thanks Tom.

closed 2021-11-14 by dh5fs