Tlf/tlf #291: tlf resets exchange message to zero (CQWW CW)

jmeloranta, 2021-11-24:
Checking the current git version for CQWW CW. It seems that the zone info is initially set correctly but after pressing return, it resets to zero. Tlf version that comes with debian keeps the zone there correctly.

zcsahok, 2021-11-24:
Thanks for checking!
Could be give a screenshot showing the issue? By _zone info_ do you mean what is shown in the bottom left corner of the scoring window?

jmeloranta, 2021-11-29:
Zone info that is in the exchanged message. I am attaching a photo. The first qso shows 00 as the message. The 2nd (in progress) shows the correct zone but then it switches to zero if I accept the qso. Photo attached:
[image]
ps. Just finished cq ww with tlf (non-git). Good work guys!

zcsahok, 2021-11-29:
OK, I see. Did a check with a freshly compiled version bd61d84 (master on Nov 24), QSOs got logged without a glitch.

[image]

What is your `git status` output?

jmeloranta, 2021-11-30:
git status says:
On branch master
Your branch is up to date with 'origin/master'.

And the latest entry in git log is:
commit 993658cc1f90ff2ab723519eb88b12bf1b7c3f4f (HEAD -> master, origin/master, origin/HEAD)
Author: Thomas Beierlein <tomjbe@gentoo.org>
Date:   Fri Nov 26 18:24:42 2021 +0100

    Fix wrong initialisation
---

To be sure, I did git pull and no new changes appeared. I also did make clean, make, make install just in case. Still the same result.

I am on Debian testing branch and using Raspberry Pi 4 (arm64).
I don't have any intel boxes to see if this is somehow platform dependent.
The tlf version from standard Debian repository works OK.

jmeloranta, 2021-12-01:
I also tried the exact version you mentioned (Nov 24; there was one commit after that). Same behavior.

zcsahok, 2021-12-01:
Looks like some architecture related issue. Checked on armv7l (32-bit) too, all OK (previously was testing on x86_64). I'm setting up a virtual aarch64 system as I have no access to Rpi4 at the moment.
Could you run `make check` just to see if all tests pass?

jmeloranta, 2021-12-01:
make check bombs with:
  CCLD     run_searchlog
/usr/bin/ld: run_searchlog.o:(.data+0x328): undefined reference to `test_ZoneFromCountry'
/usr/bin/ld: run_searchlog.o:(.data+0x350): undefined reference to `test_ZoneFromExchange'
/usr/bin/ld: run_searchlog.o:(.data+0x378): undefined reference to `test_ZoneFromLog_mixedmode'
/usr/bin/ld: run_searchlog.o:(.data+0x3a0): undefined reference to `test_ZoneFromLog'
/usr/bin/ld: ../src/searchlog.o: in function `searchlog':
/home/eloranta/c/tlf/src/searchlog.c:687: undefined reference to `get_proposed_exchange'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:915: run_searchlog] Error 1
make[2]: Leaving directory '/home/eloranta/c/tlf/test'
make[1]: *** [Makefile:1437: check-am] Error 2
make[1]: Leaving directory '/home/eloranta/c/tlf/test'
make: *** [Makefile:475: check-recursive] Error 1

zcsahok, 2021-12-01:
My tests on a virtual aarch64 system with Debian 11 were successful. Here are the steps I did. (takes ~15 mins to build. I have stock Tlf installed hence prefix/datadir point to /usr)

```
git clone http://github.com/tlf/tlf

cd tlf
autoreconf --install --force
./configure --prefix=/usr --datadir=/usr/share && make && make check
```
[image]

[image]

Then in tlf dir
```
cp share/logcfg.dat .
src/tlf -nrv
```
[image]

and

[image]

jmeloranta, 2021-12-01:
OK, I did exactly what have done. The check part compiles now but I get:
PASS: run_addcall
PASS: run_addmult
PASS: run_addpfx
PASS: run_adif
PASS: run_bands
PASS: run_cabrillo
PASS: run_checklogfile
PASS: run_clusterinfo
PASS: run_contest
PASS: run_cw_utils
PASS: run_dxcc
PASS: run_getctydata
FAIL: run_getexchange
PASS: run_initial_exchange
PASS: run_lancode
PASS: run_locator2longlat
PASS: run_logutils
PASS: run_parse_logcfg
PASS: run_prefix
PASS: run_readcalls
PASS: run_recallexchange
PASS: run_score
PASS: run_searchlog
PASS: run_sendbuf
PASS: run_workedstations
PASS: run_wwv
PASS: run_zone_nr
============================================================================
Testsuite summary for Tlf 1.5~git
============================================================================
# TOTAL: 27
# PASS:  26
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See test/test-suite.log
Please report to tlf-devel@nongnu.org
============================================================================

getexchange test fails on my system. This is Debian testing branch with everything up to date. And tlf behaves the same way (resets exchange to 00). The compiler I have is:
gcc version 11.2.0 (Debian 11.2.0-10)

zcsahok, 2021-12-01:
ok, please run the getexchange test to see where it fails.
```
test/run_getexchange
```

jmeloranta, 2021-12-01:
Here is the logfile:
[==========] Running 4 test(s).
[ RUN      ] test_getexchange_arrlss
[       OK ] test_getexchange_arrlss
[ RUN      ] test_getexchange_cqww
[  ERROR   ] --- "00" != "12"
[   LINE   ] --- test_getexchange.c:261: error: Failure!
[  FAILED  ] test_getexchange_cqww
[ RUN      ] test_getexchange_serial_section
[       OK ] test_getexchange_serial_section
[ RUN      ] test_getexchange_sectn_mult
[       OK ] test_getexchange_sectn_mult
[==========] 4 test(s) run.
[  PASSED  ] 3 test(s).
[  FAILED  ] 1 test(s), listed below:
[  FAILED  ] test_getexchange_cqww

 1 FAILED TEST(S)
FAIL run_getexchange (exit status: 1)

jmeloranta, 2021-12-01:
I am not familiar with the code but I am looking at function checkexchange_cqww() in src/getexchange.c I see:

        index = g_match_info_fetch(match_info, 1);
 
This seems to return empty string for variable index ("" with strlen = 0). I have checked that the contest of variable comment is correct (for example, 12 in the test case). So, the following if statement never succeeds (since strlen(index) = 0):

        if (index != NULL && strlen(index) >= 1 && strlen(index) <= 4) {
            zone = atoi(index);
        }
and the zone variable is left with a value of zero that it was initialized to. Something to do with the glib regexp stuff, which I am not too familiar with. This is the dev package that I have installed:
libglib2.0-dev/testing,now 2.70.1-1 arm64 [installed]

jmeloranta, 2021-12-01:
OK, more digging. Variable "index" is just fine after g_match_info_fetch() call but this seems to reset it to empty string:
        if (index_fix != NULL) {
            g_free(index);
            index = index_fix;
        }

zcsahok, 2021-12-02:
Looks like the condition here is not sufficient
```
	if (index_fix != NULL) {  //<<<<<<
	    g_free(index);
	    index = index_fix;
	}
```
Current (2.70) glib docs says it

> If match_num is a valid sub pattern but it didn’t match anything (e.g. sub pattern 1, matching “b” against “(a)?b”) then an empty string is returned.

Further checks shall cure the issue:
```
	if (index_fix != NULL && strlen(index_fix) >= 1 && strlen(index_fix) <= 4) {  //<<<<<<
	    g_free(index);
	    index = index_fix;
	}
```
I will also run a test with Debian testing.

jmeloranta, 2021-12-02:
Perfect! Thank you!!

zcsahok, 2021-12-02:
The issue is reproducible on Debian testing (bookworm).
GLib behavior was fixed to match the docs (_g_match_info_fetch_named not return empty string as expected_) in this commit

https://gitlab.gnome.org/GNOME/glib/-/merge_requests/1751

It is part of release 2.69.0
https://gitlab.gnome.org/GNOME/glib/-/releases/2.69.0

Debian stable (bullseye) has 2.66.8, whereas testing has 2.70.1

zcsahok, 2021-12-02:
Thanks for debugging!

PR #297 (2021-12-02): Correct handling of zone fix for CQWW when using GLib 2.69+
Fixes #291
closed 2021-12-02 by zcsahok