Tlf/tlf #334: Log truncated when exporting Cabrillo

ncw, 2022-08-01:
I used TLF (latest git version) for working the IOTA contest this weekend. TLF worked fine during the contest, but when I came to export my log to Cabrillo things started to go wrong.

The initial log looked like this

```
 40CW  30-Jul-22 13:02 0001  XXXX           599  599  145EU8                 0   7021.0
 20CW  30-Jul-22 13:12 0002  XXXX           599  599  087                    0  14003.6
 20CW  30-Jul-22 13:26 0003  XXXX           599  599  172EU002               0  14013.5
 20CW  30-Jul-22 13:42 0004  XXXXX          599  599  062EU029               0  14016.2
 20CW  30-Jul-22 13:47 0005  XXXX           599  599  076                    0  14018.4
```

But when I exported it to Cabrillo using the exchange "# EU-005" I got this

```
[snip header]
QSO:  7021 CW 2022-07-30 1302 M5NCW         599 0001 E XXXX          599 145EU8
QSO: 14003 CW 2022-07-30 1312 M5NCW         599 0002 E XXXX          599 087   
QSO: 14013 CW 2022-07-30 1326 M5NCW         599 0003 E XXXX          599 172EU0
QSO: 14016 CW 2022-07-30 1342 M5NCW         599 0004 E XXXXX         599 062EU0
QSO: 14018 CW 2022-07-30 1347 M5NCW         599 0005 E XXXX          599 076   
END-OF-LOG:
```

You can see here that my exchange has been truncated, but much more worringly the remote exchange has been truncated too. For instance "172EU0" should be "172EU002".

Have I misconfigured TLF somehow to cause this? Or is it a bug?

According to the RSGB site https://www.rsgbcc.org/hf/formats/templates.shtml the cabrillo log should look like this

```
QSO: 28000 CW 2002-07-27 1600 G3XTT         599 0001 EU-005 EA5BM         599 0031 ------
QSO: 28000 CW 2002-07-27 1601 G3XTT         599 0002 EU-005 PY1NX         599 0186 ------
QSO: 28000 CW 2002-07-27 1602 G3XTT         599 0003 EU-005 M6T           599 0451 EU-005
QSO: 28000 CW 2002-07-27 1611 G3XTT         599 0004 EU-005 9M6A          599 0310 OC-088
```

I don't mind a bit of post contest log massaging, but the fact that the exported log has truncated exchanges from the remote station makes that impossible so I ended up writing a python program to make the log in the correct format!

I've put a complete set of files here needed to recreate the problem: https://gist.github.com/52cbaa8da1e3bd6319323d2668781b3e

zcsahok, 2022-08-01:
Has your log file (*.log) been also truncated or changed during the export process or "just" the export was incorrect?

zcsahok, 2022-08-01:
Assuming that only the generated Cabrillo file was affected:
Here is the Cabrillo specific part of my configuration for the IOTA contest. I just added a fake island reference for demonstration.
```
CABRILLO=TESLA

CABRILLO-CONTEST= RSGB-IOTA
CABRILLO-LOCATION= WORLD
CABRILLO-CATEGORY-MODE= CW
CABRILLO-CATEGORY-BAND= ALL
CABRILLO-EXCHANGE= # EU222
CABRILLO-CATEGORY-POWER= QRP
CABRILLO-CATEGORY-TIME= 12-HOURS
CABRILLO-CATEGORY-OPERATOR = SINGLE-OP
CABRILLO-CATEGORY-STATION = -
CABRILLO-CATEGORY-TIME = -
CABRILLO-CATEGORY-TRANSMITTER = ONE
CABRILLO-CATEGORY-OVERLAY = -
CABRILLO-CLAIMED-SCORE = -
CABRILLO-CLUB = -
CABRILLO-NAME = -
CABRILLO-EMAIL = x@x.com
CABRILLO-ADDRESS = -
```
The key is using `TESLA` template instead of `UNIVERSAL`, as it defines 10-char wide exchanges (`...;EXC_S,10;...;EXCH,10`). `UNIVERSAL` fits best for contests with either a plain serial number or a short (max. 6 characters) fixed exchange. This is why in your case e.g. `172EU002` was cropped to `172EU0`.

Here's how the Cabrillo file looks like with the above settings
```
CONTEST: RSGB-IOTA
CALLSIGN: HA5CQZ
CATEGORY-ASSISTED: ASSISTED
CATEGORY-BAND: ALL
CATEGORY-MODE: CW
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-POWER: QRP
CATEGORY-TIME: 12-HOURS
CATEGORY-TRANSMITTER: ONE
LOCATION: WORLD
EMAIL: x@x.com
QSO:  7020 CW 2022-07-30 1234 HA5CQZ        599 0001 EU222 XX0DX         599 123 EU333 
```

ncw, 2022-08-03:
> The key is using `TESLA` template instead of `UNIVERSAL`, as it defines 10-char wide exchanges (`...;EXC_S,10;...;EXCH,10`). `UNIVERSAL` fits best for contests with either a plain serial number or a [...]

Yes that seems to be the key  and fixes the export - thank you :-)

10 chars isn't quite enough using the RSGB recommended format of `1234 EU-005` - it would need to be 11.

TESLA is a really bizarre thing to write though and I'd have never worked that out looking at the docs! The only place it is mentioned is in `cabrillo.fmt` and that is a pretty cryptic file.

Ideas
- An error that entries have been truncated on output would be really useful too as I almost didn't notice the exchanges had been truncated.
- Maybe the UNIVERSAL (or a different named) format should make each field wide enough for all the entries? I think that all the contest importers can take any spacing on cabrillo logs now-a-days (can't remember where I read that).
- Maybe the last field should never be truncated? This would have fixed my issue.

PS: Do you think you could submit your IOTA rules to the repo? Its really hard making new rules for newbies, and I haven't figured out how to make tlf keep score for IOTA!

zcsahok, 2022-08-03:
Cabrillo generation is described in Manual.md in subsection 'Cabrillo formatting'. It is somewhat technical and terse, so any comments are welcome. (BTW thanks for reporting this issue!)
The description file `cabrillo.fmt` is looked up in the current directory first, so users can construct their own formats.

TESLA refers to the Tesla Memorial contest (http://www.radiosport.yu1srs.org.rs/HFTeslaMemorial/index.php/rules). It uses a serial+grid exchange, probably this was the first case when one hit the 6-char exchange width limit. Which actually stems for the Cabrillo specs page (https://wwrof.org/cabrillo/cabrillo-qso-data/): the example defines 6-char wide exchanges. The very same page contains also the statement you were referring to (_All of the major contest sponsors will also accept the fields in any column as long as there is a space between each item_). 

Truncating data fields has to be regarded as an error instead of silently ignoring them as it is done (and documented) currently. I'm preparing a fix for this.
I think it would also make sense to extend UNIVERSAL to 14-char exchanges, as that is the current default exchange width.

For the scoring: TLF in the current master branch is not capable enough to support IOTA scoring. There is an ongoing development (dev_plugin branch) to support flexible scoring via contest-specific Python plugins, it just can't handle multipliers at the moment.

ncw, 2022-08-04:
> Cabrillo generation is described in Manual.md in subsection 'Cabrillo formatting'. It is somewhat technical and terse, so any comments are welcome. (BTW thanks for reporting this issue!) The [...]

Thanks for pointing those out - I don't think I've seen those before! I'm new to cabrillo files and I naively assumed that it was a standard format!

> TESLA refers to the Tesla Memorial contest (http://www.radiosport.yu1srs.org.rs/HFTeslaMemorial/index.php/rules). It uses a serial+grid exchange, probably this was the first case when one hit the [...]

Ah ha!

> Truncating data fields has to be regarded as an error instead of silently ignoring them as it is done (and documented) currently. I'm preparing a fix for this.

That is perfect. I had a skim through your PR and it looks very well done with tests too :-)

> I think it would also make sense to extend UNIVERSAL to 14-char exchanges, as that is the current default exchange width.

I saw your commit for that too. I think that would be a great idea.

> For the scoring: TLF in the current master branch is not capable enough to support IOTA scoring. There is an ongoing development (dev_plugin branch) to support flexible scoring via contest-specific Python plugins, it just can't handle multipliers at the moment.

Contest specific python plugins sounds like a great idea. Writing C code is very laborious in comparison and python is nice and easy to embed in C (I have done this before!).

Thank you for your excellent detailed replies and for fixing the root of the problem - much appreciated :-)