Tlf/tlf #346: how to contribute with new contest rule

ON2PSA, 2022-10-03:
Hi
How can I upload to the repertoire a new rule for an existing contest (UBA ON)?
Can the cty.dat be updated according to the most recent file?
Thanks.
ON2PSA

dl1jbe, 2022-10-03:
> Hi How can I upload to the repertoire a new rule for an existing contest (UBA ON)? 

You could 
a) make a clone of the repo, add the file and make a pull request
b) send the file via the mailing list or
c) paste it somewhere and tell us the link.

> Can the cty.dat be updated according to the most recent file? 

Yes, but as cty.dat changes very often, it is mostly updated on a new release. See the FAQ how to download an actual version.

73, de Tom

zcsahok, 2022-10-03:
@ON2PSA It would be great if you could make a pull request. (option a)
Just checked the rules (https://www.uba.be/en/hf/contest-rules/uba-dx-contest-rules) and they are quite involved. Probably we will have to solve some issues to fully support them.
73, Zoli ha5cqz

zcsahok, 2023-06-27:
@ON2PSA There were a number of contests added recently that can serve as an example
#383 #384 #386 #388 #391 #393
If you have a specific issue or need help, please feel free to ask.

ON2PSA, 2023-08-01:
Dear Zoltan

I have figure out most of commands to configure rules according to contest instructions.

I think important to highlight that rules needs to be place either in /usr/share/tlf/rules or inside a folder called rules in a folder where the contest data will be created. The multipliers also need to be present in the rules folder.

Here attached I share with you my rules for Portugal Day Contest and European HF Championship.

Please review the content and post in github for release and access to other users.

Thanks for developping TLF. Great software.

Happy to help on the development (I'm not a programmer however).

Kind regards

Pedro

73

ON2PSA

 
 

Sent: Tuesday, June 27, 2023 at 7:04 PM

ON2PSA, 2023-08-01:
Dear Zoltan.
Apologies. The ptday_dx has incorrect instructions.  I'll send you a new file tomorrow night.
Cheers. 
Pedro

Dear Zoltan

I have figure out most of commands to configure rules according to contest instructions.

I think important to highlight that rules needs to be place either in /usr/share/tlf/rules or inside a folder called rules in a folder where the contest data will be created. The multipliers also need to be present in the rules folder.

Here attached I share with you my rules for Portugal Day Contest and European HF Championship.

Please review the content and post in github for release and access to other users.

Thanks for developping TLF. Great software.

Happy to help on the development (I'm not a programmer however).

Kind regards

Pedro

73

ON2PSA

Sent: Tuesday, June 27, 2023 at 7:04 PM

#383 <https://github.com/Tlf/tlf/pull/383> #384 <https://github.com/Tlf/tlf/pull/384> #386 <https://github.com/Tlf/tlf/pull/386> #388 <https://github.com/Tlf/tlf/pull/388> #391 <https://github.com/Tlf/tlf/pull/391> #393 <https://github.com/Tlf/tlf/pull/393>

zcsahok, 2023-08-02:
The attachments got lost, try to include them in the text.
Did you mean this contest  https://www.portugaldaycontest.rep.pt/rules.php ?

ON2PSA, 2023-08-02:
Hi Zoltan

Thanks for your reply.

Here are the rules for ptday and ptday_dx (Portugal day contest) and European HF contest

The multi and prefixes for ptday and ptday_dx are also included.

Please check for incoerences in the instructions.

I would like to create rules for other contest I have intention to participate and thus share with TLF users.

Thus understanding rules and how apply then according to the instructions is important to debug the errors.

Thanks.

Pedro

 

 

Here are the rules for ptday:

=================

 

#########################
# Portugal day Contest  
# version 0.1 31JUL2023 
#########################
#
CONTEST=PORTUGAL DAY
LOGFILE=PTday.log
CONTEST_MODE
CABRILLO=UNIVERSAL
#
##################################
#    Rules Portugal side      #
#                 #
##################################
MULT_LIST=portugalday_multi
SECTION_MULT
DX_&_SECTIONS
MY_COUNTRY_POINTS=5    
DX_POINTS=1        
#
##################################
#                                #
#  Messages F1= to F12=          #
#  Message CQ_TU_MSG=            #
#  Message S&P_TU_MSG=           #
#                                #
#  % = call                      #
#  @ = hiscall                   #
#  # = serial                    #
#  [ = RST                       #
#  + = increase cw speed         #
#  - = decrease cw speed         #
#                                #
##################################
#
F1=TEST DE % K
F2=@ DE %
F3=@ 5NN#
F4=TU 
F5= @
F6=%
F7=@ SRI QSO B4 GL
F8=AGN
F9= ?
F10= QRZ?
F11= PSE K
F12=TEST % %
#
CQ_TU_MSG=TU % 
S&P_TU_MSG=TU UR 5NN #
#
#ALT_0=
#ALT_1=
#ALT_2=
#ALT_3=
#ALT_4=
#ALT_5=
#ALT_6=
#ALT_7=
#ALT_8=
#ALT_9=
#
#SEND_DE
#
NOB4
####### END #####################

 

 

Rules for ptday_dx

============================

 

#########################
# Portugal day Contest  #
# version 0.1 02AUG2023
#########################
#
CONTEST=PORTUGAL DAY DX
LOGFILE=PTday.log
CONTEST_MODE
CABRILLO=UNIVERSAL
#
##############################################
#   Rules DX side             
#
#   DX/CT 10 points - see prefixes
#   DX with same country and continent 1 point
#   DX with other continents 2 points
##############################################
SSBMODE
SERIAL_OR_SECTION

COUNTRYLIST=prefixes
COUNTRY_LIST_POINTS=10

MY_COUNTRY_POINTS=1
MY_CONTINENT_POINTS=1    
DX_POINTS=2        
#
##################################
#                                #
#  Messages F1= to F12=          #
#  Message CQ_TU_MSG=            #
#  Message S&P_TU_MSG=           #
#                                #
#  % = call                      #
#  @ = hiscall                   #
#  # = serial                    #
#  [ = RST                       #
#  + = increase cw speed         #
#  - = decrease cw speed         #
#                                #
##################################
#
F1=TEST DE % K
F2=@ DE %
F3=@ 5NN#
F4=TU 
F5= @
F6=%
F7=@ SRI QSO B4 GL
F8=AGN
F9= ?
F10= QRZ?
F11= PSE K
F12=TEST % %
#
CQ_TU_MSG=TU % 
S&P_TU_MSG=TU UR 5NN #
#
#ALT_0=
#ALT_1=
#ALT_2=
#ALT_3=
#ALT_4=
#ALT_5=
#ALT_6=
#ALT_7=
#ALT_8=
#ALT_9=
#
#SEND_DE
#
NOB4
####### END #####################

 

prefixes file content:

============

 

Portugal:CQ0,CQ7,CR5,CR6,CR7,CS1,CS2,CS5,CS7,CT1,CT2,CT4,CT5,CT6,CT7
Azores:CQ8,CU1,CU2,CU3,CU4,CU5,CU6,CU7,CU8,CU9,CR1,CR2,CR8,CS4,CS8,CT8
Madeira:CQ2,CQ3,CQ9,CR3,CR9,CS3,CS9,CT3,CT9

 

portugalday_multi content:

================

 

AV
BJ
BR
BG
CB
CO
EV
FR
GD
LR
LX
PG
PT
SR
ST
VC
VR
VS
CAL
CMB
FU
MC
PS
PM
PTS
RB
SCM
STM
SV
AH
CA
SCG
VL
PV
CV
HT
LJF
LJP
MD
SCF
SRP
LG
ND
PD
PO
RG
FC
VP

 
rules for European HF champ "EUROHF"

#########################
# Portugal day Contest  
# version 0.1 31JUL2023 
#########################
#
CONTEST=european_HF
LOGFILE=european_hf.log
CONTEST_MODE
CABRILLO=UNIVERSAL
#
#################################################
#    Rules 
#    One QSO one point
#    Multiplier is licence year once per band
#################################################
ONE_POINT
WYSIWYG_ONCE
#
##################################
#                                #
#  Messages F1= to F12=          #
#  Message CQ_TU_MSG=            #
#  Message S&P_TU_MSG=           #
#                                #
#  % = call                      #
#  @ = hiscall                   #
#  # = serial                    #
#  [ = RST                       #
#  + = increase cw speed         #
#  - = decrease cw speed         #
#                                #
##################################
#
F1=TEST DE % K
F2=@ DE %
F3=@ 5NN#
F4=TU 
F5= @
F6=%
F7=@ SRI QSO B4 GL
F8=AGN
F9= ?
F10= QRZ?
F11= PSE K
F12=TEST % %
#
CQ_TU_MSG=TU % 
S&P_TU_MSG=TU UR 5NN #
#
#ALT_0=
#ALT_1=
#ALT_2=
#ALT_3=
#ALT_4=
#ALT_5=
#ALT_6=
#ALT_7=
#ALT_8=
#ALT_9=
#
#SEND_DE
#
NOB4
####### END #####################

 

 
 

Sent: Wednesday, August 02, 2023 at 9:56 PM