Tlf/tlf #348: Review usage of fgets in file reading

zcsahok, 2022-10-05:
Follow-up from #345:

As the problem with long lines pops up from time to time, maybe we should check all `fgets()` invocations for truncation of input due to limit buffer size.
There is another function `getline()` which seems better fit this use case.

dl1jbe, 2022-10-07:
Scanning thru the code there are quite some places where 'fgets()' is used. 

For start it may be best to begin with user provided files (multipliers, config.dat, rules file, initial_exchange, maybe cabrillo.fmt).

darvark, 2023-07-09:
I had some free time during my vacations and I've prepared code change that contains replacement of all fgets with getline function. Tests passes, tlf seems to work without problem. Soon I'll add pull-request

zcsahok, 2023-09-09:
Fixed by #403

closed 2023-09-09 by zcsahok