Tlf/tlf #353: Handling of user provided contest rule and multiplier files

dl1jbe, 2022-10-10:
From time to time users provides contest files (rules and multipliers) for new contests. 
We may be not able to confirm the correctness of the files in short time. But it seems to be good to incorporate these contest into the distribution.

I would suggest to somehow mark these files as 'not officially' provided. The following ideas comes into mind:

- Add  an additional directory 'user_provided' into the distribution and place all relevant files there.
- Store them on the wiki (would need an additional download, but could be managed by the users themself)
- At least add some comment lines stating the files are user provided and give information about author and date

airween, 2022-10-10:
Or we should create a separated repository, eg. "tlf_unofficial_contests" or something similar, where users could push their own rules. But all of the listed items above are good.

zcsahok, 2022-10-10:
Good question.
Having them in the main repo has the advantage that one has to clone this one and also packaging has to consider only it.
A separate repo could need some change in 3rd party packaging (e.g. Debian).
On the other hand with a separate repo one would de-couple the rules from the code. But then we shall also consider migrating the current rules into this new repo. (Even the rules we have now could be out-of-date, e.g. I'm not sure that OK-OM list is still valid)
For the naming maybe something like _tlf_contests_ would be better. I find "unofficial" suggests some kind of distancing.

zcsahok, 2022-10-10:
And it would be also good to have contributed test logs to be able to check the rules. (check ok = log could be loaded without re-scoring warning)

dl1jbe, 2022-10-14:
>  Having them in the main repo has the advantage that one has to clone this one and also packaging has to consider only it. A separate repo could need some change in 3rd party packaging (e.g. Debian). 

I would underline that too.
Another point in having a separate, maybe open, repo (as @airween suggested) is the need to care for it staying clean from spam. 

> But then we shall also consider migrating the current rules into this new repo. (Even the rules we have now could be out-of-date, e.g. I'm not sure that OK-OM list is still valid)

That happens any time as contest organizers often changes their rules (LZ-DX is another example),
We should at least start to place some comment lines on start of the rules and that can be started now.

'''
 provided by:
 last verified:
'''

> For the naming maybe something like _tlf_contests_ would be better. I find "unofficial" suggests some kind of distancing.

I had suggested something like _user_provieded_, but that has the same questionable tone.

zcsahok, 2022-10-14:
One more point: we could consider separating rule files into directories per contest. So instead of using `rules/adx` move it to `rules/adx/adx` along with the other related support files (inital exchange, ...).
This would reduce the clutter in `rules`. The code could cope with this: it would have to check the subdir if the file `adx` is missing.