Tlf/tlf #356: INITIAL_EXCHANGE handling

dh5fs, 2022-10-24:
INITIAL_EXCHANGE is a nice thing for contests with a predefined exchange. It is tied together with RECALL_MULTS. This is a problem in a contest like WAG (DL side), because there are not only DOK's from the multi file are exchanged but also serial numbers from non-DL stations which should not get recalled. So both commands should not be coupled together.

zcsahok, 2022-10-25:
I see the point. Not all DL stations are in the initial exchange file so there must be some other criterion to select whether to recall an exchange or not. (btw RECALL_MULTS is a misnomer as it recalls the exchange irrespective of whether it's a mult or not)

Maybe recall of exchanges looking like a serial number could be optionally disabled?

dh5fs, 2022-10-25:
Yes, if that can be distinguished from numeric looking Mults? Now it recalls every exchange. For my use it would be fine if everything that is inside the initial exchange (if its used) file gets filled it and that's it. In the recent WAG all previous received serial numbers got recalled which was a bit annoying.