Tlf/tlf #357: Make boolean configs symmetric

zcsahok, 2022-10-25:
Currently boolean config parameters can only be set to true.
Once a parameter is set in `logcfg.dat` there is no way to set it back in the rule file.

Config handling can be easily extended in a backward compatible way:
- PARAM : sets it to true; this is the current behavior
- PARAM=true (or yes) : also sets it to true (new)
- PARAM=false (or no) : sets it to false (new)
- PARAM=none of the above : error

This change would allow us to define configs that are true by default
thus avoiding the somewhat mentally hard to grab negative logic (NOxxx like NOAUTOCQ).
Maybe we can even change the recently added NO_LEADING_ZEROS_SERIAL to LEADING_ZEROS_SERIAL, if it is not too late.

I can provide a PR for this.

dl1jbe, 2022-10-25:
Good idea. Please provide a PR. 
I had a similar feeling about the need for a setting to false for different settings. So it would be really nice to have.

N0NB, 2022-10-26:
Super good idea, Zoli.

closed 2022-11-11 by zcsahok