Tlf/tlf #359: tlf doesn't recognize folder with configuration

darvark, 2022-10-31:
Hi,
Using latest Manjaro distribution and tlf  installed from yay (AUR repository) in  I noticed that even if I create folder with logcfg.dat and rules tlf always creates "tlf" folder in $HOME and starts with default logcfg.dat. 
tlf 1.4.1.r276.g97d1db58 (latest in github).

Does anything has changed in tlf configuration so it can't read cfg file?

zcsahok, 2022-10-31:
What are the exact steps that you perform?

Here is what I get with the same version (97d1db58474) as you:
```
~$ mkdir /tmp/tlf359
~$ cd /tmp/tlf359
/tmp/tlf359$ tlf

        Welcome to tlf-1.5~git by PA0R!!

Reading country data
Reading configuration data
No logcfg.dat found. Copying default config file.
Please adapt the settings to your needs.
Reading config file: logcfg.dat
Reading contest rules file: /usr/share/tlf/rules/cqww
Reading callmaster data
Log file not found, creating new one
CW-Keyer is cwdaemon

     Call = NOCALL
     My Zone = 05     My Continent = NA

Reading logfile: cqww.log

/tmp/tlf359$ ls -a
./  ../  cqww.log  logcfg.dat  .paras  tlfmarkers
```

The is no `tlf` directory created, it must be some artifact as TLF itself does not create directories (ok, except for sound log).
When started in a directory containing no `logcfg.dat` then it is copied there from `/usr/share`. But if `logcfg.dat` already exists then it is used.

darvark, 2022-10-31:
`~ mkdir test  
~  cd test; ls 
~/test tlf 
~/test ls  
 ~/test `

`        Welcome to tlf-1.5~git by PA0R!!

Reading country data
Reading configuration data
Reading config file: logcfg.dat
Reading contest rules file: rules/cqww
Reading callmaster data
Rig model number is 2010
Rig speed is 57600
Trying to start rig control
Continue without rig control Y/(N)?
`

it uses configuration stored in $HOME/tlf/logcfg.dat not default nor that which I write to any other folder

zcsahok, 2022-10-31:
I see.
Could you execute this in the directory where TLF was built?
```
grep datadir Makefile
```

Probably it points to your $HOME.

In this case $HOME/tlf/ is used instead of /usr/share/tlf.

I normally install the distribution's stock TLF to have /usr/share/tlf populated and then make the local build using 
```
./configure --prefix=/usr --datadir=/usr/share --enable-fldigi-xmlrpc
```

(--enable-hamlib is not needed, it's a fixed dependency)

darvark, 2022-11-01:
Configuration how it's built

``` Tlf Version 1.5~git configuration:

 Prefix         /usr
 Preprocessor    
 C Compiler     gcc -Wno-error

 Package features:

 With XML RPC	true
```
sources from which tlf is built are in .cache/yay/tlf-git/src  and only definition of datadir is in PKGBUILD:49:	make prefix=$pkgdir/usr datadir=$pkgdir/usr/share install

Will try to build it manually and see if it's working

darvark, 2022-11-01:
Problem solved. I'll report bug to package maintainer, tlf works as should.

closed 2022-11-01 by darvark