Tlf/tlf #362: Using HAMLIB_KEYER makes other hamlib calls fail

jothan, 2022-11-25:
I am using the latest git tlf. When adjusting the speed up and down, some hamlib calls fail sporadically (blue status bar on the bottom). I am using hamlib via rigctld (model 2).

Looking at the code, keyer calls are made on the background thread while other rig_* calls are made on the main thread. I suspect all hamlib calls should be protected by a mutex to avoid interleaving requests.

Let me know what you think. I can make a pull request with that change if you want.

73,
Jonathan VA2XZA

zcsahok, 2022-11-25:
Thanks for the report! Most likely the cause is what you described. I can't test it as I have no rig capable of CAT keying.
@df7cb could you check this?

A PR would be surely welcome.

df7cb, 2022-11-25:
Oh that explains a regression I'm seeing now, CW keying and speed adjustment works, but frequency readout stops working after a short while. Since it worked until ~2 weeks ago, I attributed it to a hamlib upgrade I did around that time, and/or RFI, but of course the problem might well be inside tlf, and only exposed by the hamlib upgrade.

I wasn't aware of threading in tlf, so I didn't pay attention to it while coding the HAMLIB_KEYER stuff. A PR is of course welcome, I can't promise when I'll have time to look at it myself.

closed 2022-11-28 by zcsahok