Tlf/tlf #367: Hamlib version depedency?

zcsahok, 2022-11-28:
For some reason my repo's actions use Ubuntu 20 for CI build and they fail since #363.
Ubuntu 20 has Hamlib 3.3 and it provides no `rig_stop_morse` function.
Before #363 there was a warning about the missing function but the actual access was optimized away as `rig_has_stop_morse()` returned a contant false. So compilation was successful.
Now this optimization is not possible thus leading to a linking error.

We have two options -

1. Require hamlib >= 4.0
2. Guard `rig_stop_morse` access with `if HAMLIB_VERSION >= 400`

What would be your the preference?

If 1) is preferred then all Hamlib version guards could be dropped.

dl1jbe, 2022-11-29:
The problem may be introduced by the recently made changes wrt the thread safe handling of hamlib calls.

As Hamlib-4.0 came out only 2 years ago (Dec2020) I assume that there will be systems with old hamlib <4.0 around for some more time. At the moment we should keep the HAMLIB_VERSION guard.

dl1jbe, 2022-11-29:
Looking into the code it seems in the old version (before PR #363) rig_has_stop_morse() did compile to everytimes returning false.
I think the compiler could optimise out the call to rig_stop_morse() in hamlib_stop_morse() on that knowledge. I suggest to revert the patch which replaced the rig_has_stop_morse/rig_has_send_morse. with the new global variables rigstopmorse/rigstartmorse.

dl1jbe, 2022-11-29:
Right, that needs to be done anyway as someone may compile with optimizations off (-O0).

zcsahok, 2022-11-29:
Preparing a PR.

PR #368 (2022-12-20): Fix pre-v4 Hamlib support
Added CI build for Ubuntu 20 to test with Hamlib 3.3

Added guard for `rig_stop_morse()`

Renamed Hamlib morse flags to avoid any clashes with `rig_stop_morse`. Applied astyle.

Fixes #367
closed 2022-12-20 by zcsahok