Tlf/tlf #373: North American Exchange Hanling

VictorDenisov, 2023-01-13:
What would be the best way to approach NA qso party exchange?
NA qso party requires sending name and state.
I use SERIAL_OR_SECTION exchange option.
If I put name and state in the field then it doesn't understand that state is a multiplier.
If I flip the order then it drops the name field:

```
40CW  13-Jan-23 05:17 0001  AA6BT          ---  ---  MB SAM                 1         
40CW  13-Jan-23 05:17 0002  W6ESL          ---  ---  CA            CA       1         
```
Is there any way to fix it what is available. Or will it require more development?

zcsahok, 2023-01-13:
It seems that the received name is not used at all. Here are the rules: https://ncjweb.com/NAQP-Rules.pdf
So is your second QSO a valid one? If yes, then just don't log the name.  :smile: 
If the name is required, then it's somewhat analogous for example to HA-DX: one country or region is kind of "zoomed in". For this #347 provides a solution.

`SERIAL_OR_SECTION` is for the case when DX stations send a serial number. Here DX sends name only.

VictorDenisov, 2023-01-13:
Name is not used indeed, but it needs to be reported in the submitted log. I can use TLF with WYSIWYG multipliers configuration, but then the score won't be correct during the contest.

VictorDenisov, 2023-01-14:
Is there any documentation on plugins? As I understand you write plugins in python.

closed 2023-04-05 by dl1jbe