Tlf/tlf #387: Feature Request: Help me stay in my bandplan

arodland, 2023-03-27:
I've used tlf for quite a while, and I usually do contests unassisted, but this time I decided to fire up the telnet option and use the Ctrl-G option a lot. It was fun, but, much to my embarrassment, I realized that several times I had absent-mindedly grabbed a region-1 station that was transmiting on a frequency where I'm not supposed to be doing SSB, and called them back.

It would be really nice if there was a way to configure allowed frequency ranges per mode, and have tlf let you know (highlight? popup?) if a station is out-of-band. Not a lock-out (maybe you're split and tlf doesn't know it, or maybe something else is going on), just some helpful advice.

arodland, 2023-03-27:
People could also set up their config to help them stay within their license privileges.

zcsahok, 2023-03-27:
The easiest way to receive only spots matching one's band plan is to use filtering on DX cluster side: http://www.dxcluster.org/main/filtering_en-6.html

On TLF side we could add band configuration like this
```
BAND = <meters>:<low>,<cwcorner>,<ssbcorner>,<high>
```
Configuring multiple bands would require multiple such lines.

arodland, 2023-03-28:
> The easiest way to receive only spots matching one's band plan is to use filtering on DX cluster side

That's a good point, but again, I don't want to *not see them*. They might be listening on another frequency, for instance. I just want them to be conspicuously highlighted or something. If we're fancy it could be a bandmap filter also.

zcsahok, 2023-03-30:
I see. Out-of-band spots are currently simply discarded on receipt. One option would be to add a further control code for the BANDMAP config (say **X** for _extra-band_) to keep them but show marked as out-of-band. These spot would be not grabbable, neither evaluated for a multiplier. It would be up to the user to find a potential way to work them (split, cross-band, ...).

E.g. having limited 160m for 1840k and up and configured BANDMAP=BX could yield

[image]

zcsahok, 2023-04-28:
@arodland any feedback on the proposed solution?

PR #392 (2023-07-18): Make bands configurable
Aiming to resolve #387.

Bands can be specified using the `BAND_mmm` keywords, where `mmm` is the band's wavelength . The argument is a comma separated tuple of 4 frequencies: band bottom, CW corner, SSB corner, band top. Frequencies must be specified in Hz, with an optional **k** suffix.

An example for 40m according to Region 1 band plan (https://www.iaru-r1.org/on-the-air/band-plans/):
```
BAND_40 = 7000k, 7040k, 7050k, 7200k
```

Will update the man page and also add marking spots on bandmap.
closed 2023-07-18 by zcsahok