Tlf/tlf #394: Cannot build with latest (git) Hamlib

GeoBaltz, 2023-06-17:
Symbol conflict with rig.h - partial make output:
make[2]: Entering directory '/data/gwb/OSS/src/tlf/src'
  CC       addcall.o
In file included from addcall.c:47:0:
globalvars.h:107:24: error: ‘rig_lock’ redeclared as different kind of symbol
 extern pthread_mutex_t rig_lock;
                        ^~~~~~~~
In file included from tlf.h:27:0,
                 from addcall.h:25,
                 from addcall.c:39:
/usr/local/include/hamlib/rig.h:2914:1: note: previous declaration of ‘rig_lock’ was here
 rig_lock(RIG *rig, int lock);
 ^~~~~~~~
In file included from setcontest.h:25:0,
                 from addcall.c:52:
globalvars.h:107:24: error: ‘rig_lock’ redeclared as different kind of symbol
 extern pthread_mutex_t rig_lock;
                        ^~~~~~~~
In file included from tlf.h:27:0,
                 from addcall.h:25,
                 from addcall.c:39:
/usr/local/include/hamlib/rig.h:2914:1: note: previous declaration of ‘rig_lock’ was here
 rig_lock(RIG *rig, int lock);
 ^~~~~~~~
make[2]: *** [Makefile:600: addcall.o] Error 1
make[2]: Leaving directory '/data/gwb/OSS/src/tlf/src'
make[1]: *** [Makefile:484: all-recursive] Error 1
make[1]: Leaving directory '/data/gwb/OSS/src/tlf'
make: *** [Makefile:380: all] Error 2

zcsahok, 2023-06-17:
Thanks for reporting this issue. Which hamlib version do you use for building TLF?

zcsahok, 2023-06-17:
OK, found it: `rig_lock` was introduced here  https://github.com/Hamlib/Hamlib/commit/fc96ec08742ee5b9cc3592744ccbe00ee5b55a43
It's not yet in the latest release (4.5.5).
Will provide a fix for it.

PR #395 (2023-09-09): Rename rig_lock to rig_mutex to avoid clash with hamlib
Fixes #394: mutex `rig_lock` renamed to `rig_mutex` to avoid the clash.
Other mutexes are also called this way, so it also aligns the naming with the rest of the code.
closed 2023-09-09 by zcsahok