Tlf/tlf #410: CW keying via hamlib

sp7q, 2023-10-03:
CW keying via HamLib through "b" command line in this thread : 
https://www.cqrlog.com/node/1484

zcsahok, 2023-10-09:
I believe this works already (see #294). Please check the `HAMLIB_KEYER` keyword.

zcsahok, 2024-01-08:
@sp7q Is there anything open regarding Hamlib keying?

df7cb, 2024-03-22:
A new release featuring it... :)

dl1jbe, 2024-03-28:
When will be the next mayor Debian release?

N0NB, 2024-03-28:
* On 2024 28 Mar 10:38 -0500, Thomas Beierlein wrote:
> When will be the next mayor Debian release?

Probably sometime around mid 2025 as Debian 12 was released last summer.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

dl1jbe, 2024-03-29:
ok, a new release is overdue anyway.

dl1jbe, 2024-09-26:
As only a new version release is left off Lets close the issue itself.

closed 2024-09-26 by dl1jbe