Tlf/tlf #411: Submitting contact with several multipliers

VictorDenisov, 2023-10-15:
California QSO party has this concept of county lines. Contact with county line stations gives two multipliers.
Are there any ways for recording several multipliers in one QSO?

zcsahok, 2023-10-15:
If it were contest exchange + DXCC, then it would be possible with some minor coding. (see last comment for #252 )
Recording two arbitrary multipliers is currently not possible, but it's on the roadmap (that why generic multiplier is called `mult1`...). It'll however require a substantial rework of addmult.

zcsahok, 2023-11-22:
I have a possible solution in https://github.com/zcsahok/tlf/tree/poc_xmult

The idea is to treat the plugin-generated multiplier string as a list of space separated individual mults and check each of them. The ones that could been applied will be added to the mult field of the log line. There are only 8 chars available, so for two 4-char wide counties the second one gets a bit truncated.

Another basic change is to update `mult1_value` with the actually applied mults, thus moving away from using globals like  `new_mult`.

The rules are just checked in directly under test/rules/cqp_dx. After review they have to be moved to rules/. A separate test for CA stations could also be added.

zcsahok, 2023-11-23:
[image]

zcsahok, 2023-12-02:
Yes, it's only truncated in the "applied multipliers" field. The last QSO works TULA again and it is not counted, only KERN. Same for the second QSO on the screenshot.

VictorDenisov, 2024-10-01:
Are there plans to merge this change?

zcsahok, 2024-10-01:
I'm working on it.

PR #442 (2024-10-24): Support several multipliers for a QSO
Allow generating several multipliers for a single QSO. These are generated by the corresponding contest plugin as a space separated list. Usages include both explicit multipliers as for California QSO party and also derived ones as for EU-DX. In the latter case both DXCC and the EU region code count as multipliers.

Added rules for EU-DX.

Resolves #411.
closed 2024-10-24 by zcsahok