Tlf/tlf #414: Use of 'getline' (PR 403) needs more work

dl1jbe, 2023-10-23:
There seems to be some problems in the switch to using getline() instead of fgets().

a) In some places the return value is checked against -1 (error) and in other cases against +1 (one char read).

b) Reading from man page if the function does *NOT* return -1 it means that one line got read from the file without errors. So checking for the length of the buffer being enough and checking for ENOMEM inside the while-loop seems unneeded. It should be checked after leaving the loop to decide between error and EOF condition.

As a further hint - the null pointer guard 'if (buffer) ..' before free() can be dropped as free works fine with null pointers.

@darvark Do you have a chance to work on  it? I would suggest starting with one instance and using it as a template for the others later.

darvark, 2023-10-23:
Hi,
Yes I can work on it.

dl1jbe, 2023-10-23:
Good. I would suggest starting with the one in cabrillo_utils.c.

zcsahok, 2023-10-23:
Replacing getline() was a longer process. In my comment https://github.com/Tlf/tlf/pull/398#issuecomment-1665115762 I proposed a generic line-by-line file reading function. This was then postponed to a later stage (https://github.com/Tlf/tlf/pull/398#issuecomment-1670100505). I can also take it.

dl1jbe, 2023-10-23:
Yes I saw it from the mails and comments on the PRs. 
I just did a check and saw the inconsistent checks of the return value and the wrong error handling.

Your idea of a generic processing function is a good one. I would suggest that we as a first step correct the problems and migrate afterwards. Let us give @darvark a chance to sort it it out with our help.

darvark, 2023-10-25:
Now I'm preparing my club station for cqww, so I have not enought free time, please be patient, I'll start working on that just after cqww ssb.

dl1jbe, 2023-10-25:
Take your time and good luck in the contest.

darvark, 2023-12-22:
Sorry for late reply, finally I found some time and I think I manage to unify all usage of getline(), but now I have problem with one test: run_rules, and as far as I know it uses python script run_tlf.py. As soon as I solve that issue I'll push changes for review

dl1jbe, 2024-01-07:
Commit got merged. If no further comments, I will close the issue in coming week.

zcsahok, 2024-01-08:
Fine with me. I'll look into the generic reading once I get to it.

closed 2024-01-19 by dl1jbe