Tlf/tlf #418: Building without specific depedencies possible?

rwinner, 2023-12-15:
Hi, just wondering if it is possible to build from source without the hamlib, sox or xmlrpc-fldigi support. IOW, just as a stand alone basic logger for logging contacts. Thanks!

dl1jbe, 2023-12-15:
Am Fri, 15 Dec 2023 06:51:07 -0800
schrieb rwinner ***@***.***>:

> Hi, just wondering if it is possible to build from source without the
> hamlib, sox or xmlrpc-fldigi support. IOW, just as a stand alone
> basic logger for logging contacts. Thanks!
> 

Hi,

from the three mentioned above only hamlib is mandatory.

Sox is used when you run Tlf for SSB contests but it is only an
external tool and not referenced during compilation.

xmlrpc-c ist only needed to communicate with fldigi and controlled by
the '--enable-fldigi-xmlrpc' switch during the configure step.
If you leave that parameter out than Tlf should compile without
xmlrpc-c installed.

Regards,
	Tom DL1JBE
-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

rwinner, 2023-12-15:
Thanks!

Lee 😎

dl1jbe, 2024-01-14:
It seems that no more discussion is needed. Closing.

closed 2024-01-14 by dl1jbe