Tlf/tlf #422: Sporadic lock-up on exit

zcsahok, 2024-01-10:
Sometimes TLF freezes on exit. It happens quite rarely, but the other day I had a chance to look into it using gdb. The process sits in `close_tlf_rig()` waiting for `tlf_rig_mutex` (see #363) .

What happens is the following

- `background_task` performs a Hamlib operation with `tlf_rig_mutex` locked
- user exits TLF and `background_task`'s thread is instructed to terminate using `pthread_cancel()` in `tlf_cleanup()`
- the thread exits according its default setting `PTHREAD_CANCEL_DEFERRED`, that is roughly after next syscall, irrespective of the application state, including mutexes.
- so `tlf_rig_mutex` stays locked
- once the thread had exited, `close_tlf_rig()` tries to lock `tlf_rig_mutex`, but never manages it
- process has to be killed manually

As code spends just some milliseconds in Hamlib calls all this happens only occasionally, but it's still quite annoying.

The preferred way of existing `background_task` would be gracefully signaling via a shared variable. Something like
```
  while (!stop_backgrnd_process) {
       // loop body
  }
```
Current `stop_backgrnd_process` should be renamed to `pause_backgrnd_process` as it effectively only pauses/suspends it.

dl1jbe, 2024-01-11:
Good catch.

There are some other ways for a clean cancel of a thread:

a) Use `pthread_set_cancelstate` to protect the critical parts of the thread where a cancel could led to hangups. At a first glance that should be `write_keyer`, `cqww_simulator` and `gettxinfo`.

b) Install clean-up handlers to release any held mutexes in functions locking a mutex temporarily.

As the turn around time of the background loop is only some ms any of the three variants should work.

zcsahok, 2024-01-13:
Unless there is a reason I'd prefer to keep it simple. Will test the leaving background process on a shared variable and report back.

dl1jbe, 2024-01-14:
Fine. Do we need a mutex for it or simple use an atomic data type for the flag?

zcsahok, 2024-01-27:
As it is a write-once-at-end flag I'd assume no guarding is needed. The fix works so far.

PR #424 (2024-02-07): Terminate background thread gracefully
Fixes #422
closed 2024-02-07 by zcsahok