Tlf/tlf #446: Python support breaks mult map

VictorDenisov, 2024-10-16:
When mult handling is done by python plugin it breaks mult map. Invoking mult map shows an empty list of multipliers.

airween, 2024-10-16:
Could you show your config to reproduce the issue?

VictorDenisov, 2024-10-18:
Github doesn't allow me to attach python file or logcfg dat. I tried adding mult_list configuration option but it doesn't help because multipliers are processed by the python plugin and python plugin doesn't have any methods for returning full list of expected multipliers

airween, 2024-10-18:
> Github doesn't allow me to attach python file or logcfg dat.

You can copy the content of those files and paste here, into the comment field. Please enclose them by a ``` separator, like this:

```
\```
import re

# match trailing call modifier (e.g. /8, /P, /MM, /QRP, etc.)
MODIFIER_PATTERN = re.compile('/(\d|[A-Z]+)$')
...
\```
```
(Don't put the leading `\` in front of ```)

For more info please take a look at the [Github formatting](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax).

VictorDenisov, 2024-10-19:
Rules
```
########################
# General contest mode #
########################
#
CONTEST_MODE
LOGFILE=sst.log
CABRILLO=UNIVERSAL
#ONE_POINT
#SERIAL_OR_SECTION
GENERIC_MULT=BAND
MULT_LIST=sst_mults
#SECTION_MULT
NO_RST
PLUGIN_CONFIG=hello
#
##################################
#                                #
#  Messages F1= to F12=          #
#  Message CQ_TU_MSG=            #
#  Message S&P_TU_MSG=           #
#                                #
#  % = call                      #
#  @ = hiscall                   #
#  # = serial                    #
#  [ = RST                       #
#  + = increase cw speed         #
#  - = decrease cw speed         #
#                                #
##################################
#
F1=cq sst %
F2=@ DE %
F3=vic ca
F4=TU 
F5=@
F6=%
F7=@ SRI QSO B4 GL
F8=AGN
F9= ?
F10= QRZ?
F11= PSE K
F12=cq sst %
#
CQ_TU_MSG=TU %
S&P_TU_MSG=vic ca
#
#ALT_0=
ALT_1=ca
ALT_2=qth?
ALT_3=vic
ALT_4=name?
#ALT_5=
#ALT_6=
#ALT_7=
#ALT_8=
#ALT_9=
#
#SEND_DE
#
CABRILLO-QSO-FORMAT=SST
CABRILLO-EXCHANGE=vic ca
CABRILLO-CONTEST=
####### END #####################
```

VictorDenisov, 2024-10-19:
Python code:
```
"""
NAQP contest
https://ncjweb.com/NAQP-Rules.pdf
"""

countries = ['CA', 'AL', 'AK', 'AZ', 'AR', 'CO', 'CT', 'DE', 'FL', 'GA', 'HI', 'ID', 'IL', 'IN', 'IA',
'KS', 'KY', 'LA', 'ME', 'MD', 'MA', 'MI', 'MN', 'MS', 'MO', 'MT', 'NE', 'NV', 'NH', 'NJ',
'NM', 'NY', 'NC', 'ND', 'OH', 'OK', 'OR', 'PA', 'RI', 'SC', 'SD', 'TN', 'TX', 'UT', 'VT',
'VA', 'WA', 'WV', 'WI', 'WY', 'DC',
# Canada
'NB', 'NL', 'NS', 'PE', 'QC', 'ON', 'MB', 'SK', 'AB', 'BC', 'NT', 'NU', 'YT',
# Other countries
'4U1/u', '6Y', '8P', 'C6', 'CM', 'CY9', 'CY0', 'FG', 'FJ', 'FM', 'FO', 'FP', 'FS',
'HH', 'HI', 'HK0', 'HP', 'HR', 'J3', 'J6', 'J7', 'J8', 'KG4', 'KP1', 'KP2', 'KP4',
'KP5', 'OX', 'PJ5', 'PJ7', 'TG', 'TI', 'TI9', 'V2', 'V3', 'V4', 'VP2E', 'VP2M', 'VP2V',
'VP5', 'VP9', 'XE', 'XF4', 'YN', 'YS', 'YV0', 'ZF']

def score(qso):
    return 1

def check_exchange(qso):
    parts = qso.exchange.split()

    file1 = open("myfile.txt" , "a" )
    file1.write(" ".join(parts))

    file1.close()

    mult = ''
    if len(parts) == 2:
        if parts[1] in countries:
            mult = parts[1]
        else:
            mult = ''
    else:
        mult = ''
    return {'mult1_value': mult}
```

VictorDenisov, 2024-10-19:
```
RULES=sst
#
#################################
#                               #
#         LOGCFG.DAT            #
#                               #
#  Uncomment the options you    #
#  want to enable.              #
#  See 'man tlf'                #
#  for a description of the     #
#  options. You can keep diff-  #
#  erent versions for different #
#  contests. I keep separate    #
#  configuration files for      #
#  each contest. If you enable  #
#  more than 1 mutually exclu-  #
#  sive options, the last one   #
#  will be efective.            #
#                               #
#                               #
#################################
#
#CTCOMPATIBLE
#
#################################
#                               #
#   EDITOR                      #
#                               #
#################################
#
#EDITOR=nano
#EDITOR=joe
EDITOR=nvim
#EDITOR=e3
#EDITOR=mcedit
#
#################################
#                               #
#  CALL                         #
#                               #
#################################
#
CALL=N6DVS
#
#
#################################
#                               #
#  Time offset from UTC         #
#                               #
#################################
#
TIME_OFFSET=0
TIME_MASTER
#
#################################
#                               #
#  LAN PORT                     #
#                               #
#################################
#  use 'addnode' only for OTHER nodes !!
#
#ADDNODE=10.0.0.115
#ADDNODE=192.168.1.2
#
THISNODE=A
#
#LAN_DEBUG
#
#################################
[... 172 lines cut ...]
#
#################################
#                               #
#     Misc commands             #
#                               #
#################################
# No automatic 'qso before msg'
NOB4
#
# No automatic cq
#NOAUTOCQ
#
# Left/right arrows don't change band...
#NO_BANDSWITCH_ARROWKEYS
#
# Marker file for Xplanet
#MARKERS=tlfmarkers
# or for just dots
#MARKERDOTS=tlfmarkers
# or for dots+calss
#MARKERCALLS=tlfmarkers
#
#HAMLIB_KEYER
#############END#################
```

VictorDenisov, 2024-10-19:
Maybe I should clarify that python plugin doesn't break the mult list feature. This feature just doesn't work with mults handled by python plugin.

VictorDenisov, 2024-10-19:
I checked code that prints out multipliers and it looks like it only works if it's ARRL sweepstakes contest and section list is a multiplier: https://github.com/Tlf/tlf/blob/master/src/changepars.c#L763

VictorDenisov, 2024-10-19:
@zcsahok Do you have any proposals as to the best way to implement multiplier list feature for python plugin? My first thought is to add a standard method that python plugins should implement to return full list of multipliers.

zcsahok, 2024-10-20:
@VictorDenisov Thanks for bringing up this issue.
When designing the plugin concept the main focus was on the ability to provide correct scoring capability. Multiplier screen was regarded as a prio 2 feature and thus postponed for later. There are a bunch of cans of worms here, so let's just open them one at a time.

There are three flavors of multiplier summary screens:
- :MUL executes `multiplierinfo()` which handles Sweepstakes and section configuration.  All other cases result in an empty screen. This is what you were experiencing.
- Alt-M or :CTY calls `show_mults()` which after handling FOCM and CQWW falls back to `multiplierinfo()`.
    :CTY is supposed to show countries, but it does this only for FOCM and CQWW.
- Alt-Z: shows zones for CQWW. For other contests it falls back to :MUL.

For sections `multiplierinfo()` uses `strstr()` for comparison, which leads to false evaluation when a multiplier string is part of another mult.
It shows only the first 4 characters of the multiplier, which again shows false results for say NAQP that uses 5-char multipliers.
So there are enough issues to fix, even without the plugin story.

But back to the original issue: how about this solution?
Plugin would provide an optional function `get_multiplier_info()` that returns a (potentially multi-line) string.
This string serves either as just a list of all mults to be handled by the legacy section code part,
or it contains a sort of a simple template that is interpreted and then shown to the user.

The two cases are -
- no formatting ('{' char) detected:
    - interpret the value as a whitespace separated list of multipliers and show them in the legacy format
    - note: mults are not sorted, they are shown in the order specified
- else:
    {@N} - evaluate further mults on band N only (default: any band), no output is generated
    {ABC} - if not yet worked according the current band selection then show the value, else show as dots.
          - for the "any band" evaluation the legacy 6-char */- string is appended
    - center the whole multi-line string vertically and per line horizontally
    - if text doesn't fit vertically on one screen then allow scrolling
    - lines longer than 80 chars are silently truncated

This would allow to specify for a 2-band contest the info like this:
```
Band A B C
40m{@40}  {A} {B} {C}
80m{@80}  {A} {B} {C}
```

with the resulting output if B has been worked on both bands and C only on 80:
```
   Band A B C
   40m  A . C
   80m  A . .
```

VictorDenisov, 2024-10-26:
Yes. `get_multiplier_info()` makes sense to me. Can you clarify the second format. Is this exact format that you propose:
```
Band A B C
40m{@40}  {A} {B} {C}
80m{@80}  {A} {B} {C}
```
And we detect if this format is used by checking if {} symbols are present in the returned string at all?

zcsahok, 2024-10-27:
Yes, that woud be the multiline-string returned by `get_multiplier_info()`. The easiest way to recognize if it's a template string is just look for '{'. Of course, then '{' shall not be used in plain strings. But as the plain strings contain just the multipliers this is not an issue.

zcsahok, 2025-01-07:
https://github.com/zcsahok/tlf/tree/poc_show_mults has the first draft of `get_multiplier_info()` support.

Added examples for NAQP and MWC.

* NAQP uses the simple interface by providing the list of multipliers. The use of some basic formatting (line breaks and alignment) is also demonstrated.
[image]

* MWC shows the template approach as this contest runs on 2 bands only.
[image]

df7cb, 2025-01-08:
The multiplier markings in the bandmap also ignore the multiplier info from python.