Tlf/tlf #454: Kenwood TS-480 rig control issue.

MikePavol, 2025-01-10:
Hello,
 Let me explain a problem with connection TS-480SAT to tlf I faced.
I have two laptops with Linux on them: AntiX and Linux Light.
There are two transceivers: Kenwood TS-480SAT and Xeigu G90, as well.
There are also usb rs232 adapters of various types: FTDI, Prolific, CP2102.
G90 works without any problem with tlf, CQRLOG and rigctl on both systems.
TS-480 works OK with CQRLOG and rigctl , but when I try to change rig from 3088 to 2028 in tlf with any combinations of port speed, have no luck.
With CQRLOG TS-480 works on 9600 port speed only.
One of the laptops has both loggers installed, the other one has only tlf.
Of course, I've checked port numbers and so on every connection.
Please, help me with at least a direction to go.
Thank you in advance,
Mike, VA3CY.

dl1jbe, 2025-01-10:
Hi Mike,

I have a TS480SAT myself and did some experiments to use it with TLF
some weeks ago. If I remember correctly besides some problems about
finding the right serial cable it worked flawlessly.

It needed a straight serial line cable with RTS/CTS connected besides
RXD/TXD. But as soon as I got rigctl working with it it worked also
with TLF.

Can you give us some feedback how you configured rigctl to work with
the rig and please also the relevant menu configurations for your TS480?

I will try to retest it on my side over the next days.

73, de Tom 

 Am Fri, 10 Jan 2025 05:32:41 -0800
schrieb MikePavol ***@***.***>:

> Hello,
>  Let me explain a problem with connection TS-480SAT to tlf I faced.
> I have two laptops with Linux on them: AntiX and Linux Light.
> There are two transceivers: Kenwood TS-480SAT and Xeigu G90, [...]

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

MikePavol, 2025-01-10:
Thank you for the quick reply! I'll do it a bit later today.

dl1jbe, 2025-01-10:
Am Fri, 10 Jan 2025 10:40:55 -0800
schrieb MikePavol ***@***.***>:

> Thank you for the quick reply! I'll do it a bit later today.
> 

Just double checked in meantime: 

- works with direct serial line and also via USB.
- tested with 9600, 19200 and 57600 baud.
- tested frequency in/output from/to TLF, switch of bands and mode from
  TLF

Please check the menuitem after the baudrate - "dts polarity". It
should be set to on. And do not forget to switch the trx off and back
on after menu changes.

73, de Tom
-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

MikePavol, 2025-01-10:
I checked again:
FTDI direct line, RTS and CTS are jumpered in DB9(7,8).
Baud rate is 9600 in rigctl, on the ts480 side was set any baud: from 4800 to 57600, it works OK with all.
Tested frequency in/output from/to rigctl, switch of bands and mode from rigctl is OK.
With "dts polarity" works in both cases ON or OFF with rigctl.
TLF 1.4.1
Rignumber is 2028
Rig speed is 9600
Trying to start rig ctrl
rig open: error

dl1jbe, 2025-01-11:
Am Fri, 10 Jan 2025 15:47:10 -0800
schrieb MikePavol ***@***.***>:

> I checked again:
> FTDI direct line, RTS and CTS are jumpered in DB9(7,8).
> Baud rate is 9600 in rigctl, on the ts480 side was set any baud: from
> 4800 to 57600, it works OK with all. Tested frequency [...]

That does not work for me. If I set the trx for one baud rate I have to
use the same from rigctl. Weird.

> TLF 1.4.1
> Rignumber is 2028
> Rig speed is 9600
> Trying to start rig ctrl
> rig open: error

Do you start both programs (rigctl and tlf) as normal user? Otherwise
the access control settings for the serial port may cause trouble.

Second idea: Is there any other program already working with the rig
(rigctld or similar)?

Btw, I see you run TLF-1.4.1. You may try the actual code from the
github repository. There are quite some new features, but regards the
serial port there should be no difference (checked it here myself).

73, Tom

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--

MikePavol, 2025-01-11:
Thank you for ideas, Tom. I'll try to play with port access and install the new version TLF. 73!

MikePavol, 2025-01-11:
I coped with it. Thank you one more time, Tom!
Just builded from source the new version , that's it. 9600 is the highest baud rate, but it's OK.
 Now works like a charm :))
Mike, VA3CY. 73!